epics-modules/asyn #12: Implement asynOption in drvAsynIPPort

MarkRivers, 2016-02-06:
I want to implement the asynOption interface for drvAsynIPPort.  Right now there are 2 options I want to implement:

`"closeOnReadTimeout"  "Y" or "N"`
In R4-27 this was added via the userFlags in the configuration command.  I don’t like that implementation because this is something one might to turn off and on at runtime.

`"hostInfo"  "New IP host info string" ` 
This will allow one to connect the same asyn port to a different IP address at runtime.  SLAC has requested this for when a device fails and they want to swap in a new one without rebooting.

This is pretty straightforward except if the hostInfo includes COM.  In that case the driver uses an interposeInterface (asynInterposeCom.c) to implement the asynOption interface.  I think this is OK, except that when it finds an unknown key it prints an error message.  I think all I need to do is to change that code so if it finds an unknown key it calls the underlying drvAsynIPPort->getOption or setOption.  Does that seem correct?

tboegi, 2016-02-06:
Hej Mark,
I'm happy to review & test your implementation.
/Torsten

MarkRivers, 2016-02-07:
This branch is ready for testing on branch drvIP_option and pull request #13.

MarkRivers, 2016-02-13:
I have merged the drvIP_option branch into master.  I am ready to release R4-29 as soon as Scott reports that the asynOption changes do what he wants, and we get the modified and new adl files converted to [edl, opi, ui] files.  Please test if possible before I release.

closed 2016-02-13 by MarkRivers