epics-modules/asyn #24: Initial value of output records with devAsynOctet.c

MarkRivers, 2016-06-01:
devAsynOctet.c does not do a read from the driver in init_record to attempt to get the initial value of output records.  The numeric device support (devAsynInt32.c, devAsynUInt32Digital.c, devAsynFloat64.c) does do this.  This should probably be added to devAsynOctet.c.

hdante, 2016-06-01:
https://github.com/epics-modules/asyn/pull/25

MarkRivers, 2016-06-02:
I have created a new branch (asynOctet_fix) that fixes this problem.  Today I found the problem with initializing waveform records; I was not setting the NORD field.  I also added pasynOctetSyncIO->disconnect(), to prevent resource leak.  testErrorsApp now works correctly; the string output records initially have the value "0.0" which is set in the constructor.

MarkRivers, 2016-06-02:
The previous version of this fix had some problems:
-  The existing asynDriver documentation says that this behavior only applies to register based interfaces, not message based interfaces.

> ********
**Initial values of output records**
The device support for output records on register based interfaces (bo, mbbo, ao, longout) does an initial read() of the value from the driver in init_record. If this read() returns asynSuccess then the record value is set to the value returned from read(). If read() returns anything other than asynSuccess then the record value is not modified.

- If the record is connected directly to a drvAsynIPPort or drvAsynSerial port it will actually try to read from the device over the wire.  This is almost certain to cause problems, such as long timeouts, etc.

I believe this means reading the initial value from the driver in init_record should _not_ be the default.  However, by making it optional using an info tag we can add this capability.

I have added a new info tag asyn:INITIAL_READBACK to enable this behavior.  This is different from the existing asyn:READBACK info tag which means that device support subscribes to callbacks for an output record, and updates the record value whenever there is a callback.  The new tag only applies to the one-time read in device support init_record.

I have modified the asynOctet_fix branch, and make a new pull request.

MarkRivers, 2016-08-14:
Fixed in #27

closed 2016-08-14 by MarkRivers