epics-modules/asyn #35: avoid static param table size?

mdavidsaver, 2016-12-13:
Is there any interest in making redundant the paramTableSize argument of the asynPortDriver ctor?  I have found the FIRST/LAST macro pattern used to calculate it is error prone.  It seems to me that it would be a simple matter to change paramList::createParam() to resize its internal storage as new parameters are added.

I am not proposing to remove the paramTableSize argument as this would break too much, just to ignore it (or maybe warn).  Only private data members should change, so existing code would continue to work as before.

MarkRivers, 2016-12-14:
That seems like a good idea.  When I wrote asynPortDriver it was my first C++ project, and I was not aware of the nice things in std:: that can make appending to lists easy, etc.

Last week I was also thinking that we should change the asynPortDriver parameter library to use std::string for string parameters, so that we don't need to pass a maximum length to getStringParam.  The existing method which passes a char * would not be changed, but would be supplemented with a method that gets an  std::string without needing to specify a maximum size.  There are limitations in paramAttribute in ADCore that come from this limitation in getStringParam.

mp49, 2016-12-15:
Any idea how using std::string or STL stuff would affect compatibility with other platforms like VxWorks? This is not so much a concern for areaDetector, but more of a concern for asynPortDriver.

mdavidsaver, 2016-12-15:
I don't think this usage will be a problem, even for vxworks 5.5 (my worst case for C++ compatibility).  In any event, APS jenkins should catch a compile failure https://jenkins.aps.anl.gov/job/epics-asyn-vx55/ and a sufficiently motivated person could run the new asynPortDriverTest to be certain (I have no access to vxworks).

MarkRivers, 2016-12-19:
I merged #36  which addresses this problem.  Note that when I tried to build it failed because of the order of libraries in asynPortDriver/unittest/Makefile.  I fixed that and pushed back to github.  I have built on linux-x86_64
linux-x86
vxWorks-ppc32 (6.9)
linux-x86.win32-x86-mingw 
linux-x86.win32-x86-mingw-static
windows-x64-static (VS 2010 and VS 2015)

MarkRivers, 2016-12-19:
Note that this fix was intended to reduce the errors that come from needed to count the number of parameters.  That was typically done by defining constants for the FIRST_XXX_PARAM and LAST_XXX_PARAM and then using address math to compute the size.  However, it is still necessary in many cases to define FIRST_XXX_PARAM, because it needs to be used to figure out when to call the base class method for writeInt32, writeFloat64, etc. as shown here:
```
    int NDDxpCollectMode;                   /** < Change mapping mode (0=mca; 1=spectra mapping; 2=sca mapping) (int32 read/write) addr: all/any */
    #define FIRST_DXP_PARAM NDDxpCollectMode
...
    else if ((function == NDDxpNumSCAs)    ||
             ((function >= NDDxpSCALow[0]) &&
              (function <= NDDxpSCAHigh[DXP_MAX_SCAS-1]))) 
    {
        this->setSCAs(pasynUser, addr);
    }
            CALLHANDEL(xiaSaveSystem("handel_ini", fileName), "xiaSaveSystem(handel_ini, fileName)");
            /* Set the save command back to 0 */
            setIntegerParam(addr, NDDxpSaveSystem, 0);
        }
    }
    else if (function < FIRST_DXP_PARAM) {
        status = asynNDArrayDriver::writeInt32(pasynUser, value);
    } 
```
It would be nice to find a cleaner way to handle this as well.

BTW I am working on a driver right now where the actual number of parameters cannot be known at compile time, but only at run-time.  Michael's fix works nicely for this use case.

MarkRivers, 2016-12-19:
> I don't think this usage will be a problem, even for vxworks 5.5 (my worst case for C++ compatibility). In any event, APS jenkins should catch a compile failure [...]

In fact the Jenkins build for vxWorks 5.5 did fail.  This is the message.  I think we need to make asyn work on vxWorks 5.5 because that is what they are still using at Diamond and some APS beamlines.

```
/usr/local/vw/tornado22-ppc/host/x86-linux/bin/ccppc -c             -DCPU=PPC604  -DvxWorks=vxWorks -include /usr/local/vw/tornado22-ppc/target/h/vxWorks.h    -O2   -Wall      -mcpu=604 -mstrict-align -mno-implicit-fp -mlongcall    -fno-builtin  -MMD -I. -I../O.Common -I. -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/os/vxWorks -I../../include  -I/Jenkins/jenkins/jobs/epics-ipac/workspace/include  -I/Jenkins/jenkins/jobs/epics-seq/workspace/include -I/Jenkins/jenkins/jobs/epics-base-3.14/workspace/include/os/vxWorks -I/Jenkins/jenkins/jobs/epics-base-3.14/workspace/include       -I/usr/local/vw/tornado22-ppc/target/h ../../asyn/asynPortDriver/asynPortDriver.cpp 
../../asyn/asynPortDriver/asynPortDriver.cpp: In method `enum asynStatus asynPortDriver::createParam(int, const char *, asynParamType, int *)':
../../asyn/asynPortDriver/asynPortDriver.cpp:841: no matching function for call to `vector<paramList *,allocator<paramList *> >::at (int &)'
../../asyn/asynPortDriver/asynPortDriver.cpp:838: warning: `enum asynStatus status' might be used uninitialized in this function
make[2]: *** [asynPortDriver.o] Error 1
make[2]: Leaving directory `/Jenkins/jenkins/jobs/epics-asyn/workspace/asyn/O.vxWorks-ppc604_long'
make[1]: *** [vxWorks-ppc604_long] Error 2
make[1]: Leaving directory `/Jenkins/jenkins/jobs/epics-asyn/workspace/asyn'
make: *** [asyn.vxWorks-ppc604_long] Error 2

MarkRivers, 2016-12-20:
I have fixed this as follows:
```
diff --git a/asyn/asynPortDriver/asynPortDriver.cpp b/asyn/asynPortDriver/asynPortDriver.cpp
index 6e88013..7118ca1 100644
--- a/asyn/asynPortDriver/asynPortDriver.cpp
+++ b/asyn/asynPortDriver/asynPortDriver.cpp
@@ -838,7 +838,7 @@ asynStatus asynPortDriver::createParam(int list, const char *name, asynParamType
     asynStatus status;
     static const char *functionName = "createParam";
     
-    status = this->params.at(list)->createParam(name, type, index);
+    status = this->params[list]->createParam(name, type, index);
     if (status == asynParamAlreadyExists) {
```
Using the [list] syntax rather than .at(list) is supported in vxWorks 5.5.  The advantage of using .at(list) is that it will throw an exception if list is out of bounds.  But there are many other places where bounds checking is not being done.

closed 2016-12-20 by MarkRivers