epics-modules/asyn #46: Reporting that the parameter value has changed to the driver

hinxx, 2017-02-08:
I have a proposal for an enhancement.

I'm working with hardware that needs to have parameters updated at certain point in time. For example if DSP on FPGA is in 'armed' mode and processing is taking place parameter updates, that involve setting a register value, need to be delayed to the time after the DSP processing has finished.

I was thinking that this might be an addition to the asynPortDriver, more specifically to the paramList class by introducing 'value changed' status. It would function in a similar manner as 'flags' member currently works. Upon PV hence param value change, 'value changed' status is set. Then asynPortDriver derived class can check if param 'value changed' status is set and can act accordingly; use the parameter value when needed (update the FPGA register when safe). Afterwards it would reset the 'value changed' status.

asynPortDriver derived class is in total control of the 'value changed' status, because it needs to knows and might even ignore it if it wants to..

Does this make sense?

hinxx, 2017-02-15:
Here is the initial stab at this - #47

hinxx, 2017-02-18:
Second attempt can be found here - #48.

MarkRivers, 2018-01-26:
Hinko found another solution to this problem was found and said he no longer needs this change.

closed 2018-01-26 by MarkRivers