epics-modules/asyn #50: Probable bug in drvVxi11.c

bfrk, 2017-03-02:
gcc warns me:

```
../../asyn/vxi11/drvVxi11.c:1714:8: warning: suggest parentheses around operand of ‘!’ or change ‘&’ to ‘&&’ or ‘!’ to ‘~’ [-Wparentheses]
     if(!flags & FLAG_NO_SRQ) pvxiPort->hasSRQ = TRUE;
```

and I guess it got it right here:  `if(!(flags & FLAG_NO_SRQ))` is most probably what the author wanted to write. (I have been bitten by the idiotic precedences for bitwise operators in the past...)

Moral of the story: too many warnings, so nobody notices the interesting ones.

norumwe12, 2017-03-02:
Mea Culpa.   
Sorry about that.

MarkRivers, 2019-03-18:
This was fixed in 7550ea670cd26cc4d55a9eb3877112c804454d6e, which is in R4-34 and later.

closed 2019-03-18 by MarkRivers