epics-modules/asyn #58: New generic testing driver request

MarkRivers, 2017-07-31:
Ralph Lange has suggested a new generic driver that can be used for testing databases in the absence of a real device.  He proposed:

> We are using the standard ASYN EPICS support, so that a dummy emulation port driver might work as an emulation replacement for *any* asyn based driver, which would make the approach very efficient.

My question:
> Can you explain how you see such a generic driver working. How would you set the new value of an input record? How would that message get to the driver?

Ralph's answer:
> (In first iteration) I would make that dummy driver as simple as possible:
> - Writes do nothing.
> - Reads do not update the record, set the timestamp from the IOC clock (ignoring TSE) and set STAT/SEVR to SIMM//NO_ALARM.
> - I/O Intr records are processed every dd seconds (per port configuration option).
>
> Aim is that everything is "green" for the user (while the special mode is still visible in the STAT of all records), the IOC console is reasonably quiet, and they can run tests (programmatically or manually) by writing to the VAL fields of input records.

My question:
> What value does the driver pass to device support for I/O Intr records?

Ralph's answer:
> The driver should not set a value. The user will write their test value from above (CA or DB), the driver must set STAT, the timestamp and trigger processing.

MarkRivers, 2017-07-31:
I think I can see how this could be implemented using asynPortDriver.  The complexity comes in writing a generic driver that does not know what drvUser strings will be defined and what asyn interface those drvUser strings will be used on.

The driver can implement the drvUser methods so it creates a list of the drvUser strings.  When it gets a call on one of the interfaces using that drvUser value it then adds it to the parameter library.

When it does callbacks to device support it needs to (optionally?) set a flag that indicates that the callback value should be ignored.

ralphlange, 2017-08-01:
A bit more context...

In our case, the ASYN based driver connects to Siemens PLCs. The use cases driving this request are:

1. Development (or update) of GUI and EPICS application while the hardware (PLC) is not available.
2. Acceptance Testing: being able to manually flip all input bits and values to implement test cases and verify GUI connection, alarm server and archiving configuration.

There are serious issues with the existing EPICS simulation mechanism: e.g. "I/O Intr" records are not processed, TSE=-2 (time stamp from device) records get invalid time stamps.

When not using the EPICS simulation mechanism, such an "emulation mode" has to be done in the driver layer. Classic EPICS device support can easily do this. For ASYN based drivers, a generic solution should be achieved.

ralphlange, 2017-08-01:
In the end it might be easier to implement this in ASYN's EPICS Device Support layer. At that place there is full access to the record, and things like periodically processing the "I/O Intr" records and setting time stamps are easier to do.

ralphlange, 2017-08-01:
Activation and configuration (e.g. period for "I/O Intr" processing, do SVAL->VAL or leave VAL untouched) should be by port and could be done through specifically named variables.

ralphlange, 2017-12-08:
Simulation mode improvements are part of EPICS Base now. Closing the issue.

closed 2017-12-08 by ralphlange