epics-modules/asyn #82: Probable bug in 'drvAsynIPServerPort', function 'readIt()'

jafonso, 2019-03-29:
Hello,

I am using 'drvAsynIPServerPort' to read binary data from a UDP/IP packet into a structure, and I noticed that the function 'readIt()' is not reading the last byte.

This may be intentional, but if not, the issue seems to be with lines 195-197 from drvAsynIPServerPort.c:

`for (x = 0; x < (int)maxchars - 1; x++) {
            data[x] = tty->UDPbuffer[x + tty->UDPbufferPos];
}`

With this implementation, if I ask it to fill a structure of exactly N bytes, it will only fill N-1.
Considering this, I propose modifying 'maxchar-1' to 'maxchar'.

I don't think it will mess up with string data, because the termination character will always be added later (lines 231-232), if there is enough space.

`if (thisRead < (int) maxchars)
data[thisRead] = 0;`

In addition, I think the remaining size of tty->UDPbuffer[] should be checked before the reading of maxchar characters, otherwise it seems to me that there is nothing stopping a client from reading beyond the end of the array.

MarkRivers, 2019-10-18:
> In addition, I think the remaining size of tty->UDPbuffer[] should be checked before the reading of maxchar characters, otherwise it seems to me that there is nothing stopping a client from reading beyond the end of the array.

Can you add the code to do this, test it, and create a pull request? Thanks.

MarkRivers, 2025-06-28:
I am trying to use this driver now and I am also seeing the behavior with it returning 1 less byte than it should.  I think I actually see 4 issues with this driver:
1) It is using maxchars-1 rather than maxchars, so it returns 1 fewer bytes than it should. https://github.com/epics-modules/asyn/blob/e417caf2c4a4332e682635c67bd32a0c4d964fc3/asyn/drvAsynSerial/drvAsynIPServerPort.c#L196
2) If there is no data available when readIt is called then it simply sleeps for 1 ms and sets thisRead=0 https://github.com/epics-modules/asyn/blob/e417caf2c4a4332e682635c67bd32a0c4d964fc3/asyn/drvAsynSerial/drvAsynIPServerPort.c#L192 and returns asynSuccess and *nbytesTransfered=0.  It seems to me that it should loop for the specified timeout waiting for the connectionListener to fill the buffer before returning nRead=0 if there is no data to read.  If the data does not arrive then it should return asynTimeout, not asynSuccess.
3) It is not checking the number of characters actually in the buffer, but rather always copies maxchars characters which can lead to reading beyond the end of the internal buffer, which can lead to an access violation. https://github.com/epics-modules/asyn/blob/e417caf2c4a4332e682635c67bd32a0c4d964fc3/asyn/drvAsynSerial/drvAsynIPServerPort.c#L196
4) The buffer is being filled in the connectionListener thread, which uses the tty->UDPbufferPos variable and modifies the tty->UDPbufferSize variables here: https://github.com/epics-modules/asyn/blob/e417caf2c4a4332e682635c67bd32a0c4d964fc3/asyn/drvAsynSerial/drvAsynIPServerPort.c#L311  The buffer is being read in the readIt() method which runs in the port driver thread, and which modifes tty->UDPbufferPos and tty->UDPbufferSize variables here: https://github.com/epics-modules/asyn/blob/e417caf2c4a4332e682635c67bd32a0c4d964fc3/asyn/drvAsynSerial/drvAsynIPServerPort.c#L200 However, there is no mutex to protect the tty->UDPbufferPos and tty->UDPbufferSize variables.  This seems like a serious issue, which is likely to cause corruption?