epics-modules/asyn #86: Add support for new 64-bit integer types

MarkRivers, 2019-06-21:
EPICS 3.16.1 and 7.0 introduced support for epicsInt64 and epicsUInt64 types (Channel Access converts them to DBF_DOUBLE).

We need to add the following to asyn:
- New asynInt64 and asynUInt64 interfaces
- New device support for 3 records:
  - epicsInt64in
  - epicsInt64out
  - waveform
- Support in asynPortDriver

Question: What's the best way to handle building on older versions of base that don't have 64-bit support?  Perhaps just use the epicsInt32 and epicsUInt32 types if DBR_INT64 is not defined?

ralphlange, 2019-06-21:
Inside C/C++ code, I would say #ifdef'ing out based on numeric EPICS version is appropriate.

For .dbd files, the snippet mechanism does a pretty good job of composing a file using different snippets (again, ifdef'd on EPICS version in the Makefile).
Wait: snippets are available since 3.15.3 - how far back does the asyn compatibility reach? Is building on 3.14 required?

MarkRivers, 2019-06-21:
3.14 is required since they are still using that at Diamond and APS accelerator I believe.  asyn currently has no dependencies beyond 3.14.

ralphlange, 2019-06-21:
No snippets, then. Copying/renaming of .dbd files for different versions instead.
For code, the numeric EPICS version defines can be added in the header (~5 lines).

anjohnson, 2019-06-21:
You don't need the snippets mechanism to construct a DBD file out of parts, this works on both 3.14 and 3.15, although GNUmake doesn't know the dependencies properly if the conditions change:
```makefile
DBD += constructed.dbd
ifdef X
  constructed_DBD += x.dbd
endif
ifdef Y
  constructed_DBD += y.dbd
endif
constructed_DBD += z.dbd
```
For C/C++ conditionals you can also pass in your own macros from the Makefile to avoid having to add the version macros. This also ensures that the Makefile and the source code turn on and off together:
```makefile
ifdef X
  USR_CFLAGS += -DBUILD_X
endif
```
etc.

ralphlange, 2019-06-22:
IIRC, setting DBD will expand the .dbd file, which you don't want in the case of a support module .dbd - that's why I decided to use snippets in a recent use case.
(I just want to add additional lines for the additional int64{in,out} device support to the .dbd if EPICS is 3.16.1 or newer.)

MarkRivers, 2019-10-16:
I have pushed a new version with these features:
- New asynInt64 and asynInt64Array interfaces.  
  - These are independent of the EPICS base version.  They do require that the type epicsInt64 be defined.  epicsTypes.h does not define epicsInt64 if __STDC_VERSION__ < 199901L.  I believe Andrew said all current compilers to support "long long" and it is always 64-bit, so I added a typedef in asynDriver.h to define it on older compilers.
  - I did not add asynUInt64.  We only have asynUInt32Digital, i.e. with a mask.  Do we need asynUInt64 or asynUInt64Digital?

- Support in asynPortDriver for new asynInt64 parameter type and asynInt64Array type.
  - As with the interfaces above these should be independent of EPICS base version, but require epicsInt64 type to be defined.

- New Int64 device support for 3 records:
  - int64in
  - int64out
  - waveform
    - This device support has been added.  It is only built on EPICS base 7 and 3.16.x (x!=0).

I did not need to use any conditionals in C/C++ code.  Either a file is built or not built, so the logic is all in the Makefile.

Thanks to Andrew's help I used the DBDCAT mechanism to build devEpics.dbd at build-time.  Previously this was a static file in asyn/devEpics.  It includes the Int64 device support only on the appropriate base versions.

However, I just discovered that DBDCAT was only added in 3.15.0.2, so the build fails on 3.14, i.e. the APS Jenkins build server.  I think I need to add the equivalent rule to the Makefile so that it can build on 3.14.

MarkRivers, 2019-10-17:
These issues have now been resolved.  The master branch of asyn now builds on both 3.14 and 7.0.

I have used this version to implement Int64 support in areaDetector, and it is also working.

The asyn unittest now does a simple test of asynInt64 with asynPortDriver.

I will add some more tests to the asyn test IOCs.

I encourage others to try it out as well.

ralphlange, 2019-10-17:
Looks good!
One of our contractors - developing asyn based drivers for data acquisition systems - had been asking for 64bit interfaces in asyn. I asked them if they see a use case for unsigned int64 as well. Will post the answer once I get it. I don't know when they will be able to test it.

MarkRivers, 2019-10-17:
Unsigned int64 is also supported now in both asyn and areaDetector.

Note that the asyn interfaces and virtually everything in areaDetector are independent of the version of EPICS base, because they are not using the new int64 records and waveform support that were added in 3.16.1.  They only require a compiler that understands "long long" and "unsigned long long".  Only the int64in/out and int64 waveform device support require 3.16.1+.

ralphlange, 2019-10-18:
Great! They just answered saying they would appreciate supporting unsigned int64.
Thanks a lot!!

MarkRivers, 2019-10-18:
This is now complete.  The testErrors application now tests it.

closed 2019-10-18 by MarkRivers