epics-modules/asyn #87: server code or client code?

vlsireddy, 2019-08-17:
Hi,
is your vx11 code is client code ?
do you have server code ?
possible to use your vxi11 API's to write server code ?

Thanks

MarkRivers, 2019-08-17:
The vxi11 code in asyn is client code, to talk to servers like oscilloscopes, etc.

It cannot be used to write server code.

vlsireddy, 2019-08-20:
Hi Mark,
             Thanks for your reply.

Background: - 
 [1] i have a complete wireshark capture between vxi 11 client and vxi 11 server and i can use that as reference [to construct state machine in server side]. i Wanted my effort to be minimal.

             Based on your experience+expertise, plz let me know, where to start writing vxi 11 server code.
few options i noted
[1] Use rpcgen to generate _svc.c or _server.c and start modifying them.
[2]  use an existing rpclib and start writing.

MarkRivers, 2020-01-03:
I don't know enough about vxi11 to answer this question.  Perhaps @norumwe12 can answer it?

norumwe12, 2020-01-03:
The ASYN VXI-11 code is for client applications.  It provides no support for servers.

closed 2020-09-19 by MarkRivers