epics-modules/asyn #93: std::vector index overrun issue in asynPortDriver.cpp

pheest, 2019-10-21:
There is an issue with paramList::flags, I experience this causing an assertion within a debug build while using VS2017.

The variable is accessed (in a threaded manner) from paramList::setFlag and paramList::callCallbacks.

I understand that the variable is used as a notifier that a particular parameter has been changed, so that callbacks need to be invoked.

There are really 2 issues:
a. I think std::vector is not the optimal STL container for this purpose.
b. More importantly, STL containers are not thread-safe - this usage needs to be protected by a mutex.

I will be able to suggest an alternate solution.

Screen shot attached (while using the simDetector):
[image]

MarkRivers, 2019-10-21:
> b. More importantly, STL containers are not thread-safe - this usage needs to be protected by a mutex.

The entire parameter library in asynPortDriver is protected by a mutex.  It is in the private member variables in asynPortDriver:
```
private:
    std::vector<paramList*> params;
    epicsMutexId mutexId;
```
Code that accesses the parameter library must call asynPortDriver.lock() before accessing it and asynPortDriver.unlock() after accessing it.  This is handled automatically by the asynPortDriver base class for all derived class methods like writeInt32(), writeFloat64(), etc.  The only time the derived class normally needs to call lock() and unlock() is in threads directly created by the derived class.

Perhaps your derived class is creating a thread and not protecting access to the parameter library with lock() and unlock().

Can you send the stack trace when you get the assertion?

pheest, 2019-10-21:
Hi Mark,
See the attached stack trace screen shot.
[image]

I'm currently working with the simDetector. The fault ocurrs when I enable image acquisition.
The error can ocurr in either callCallbacks, or in setFlag. It's reproducible.

MarkRivers, 2019-10-21:
What versions of asyn, ADCore, and ADSimDetector are you using?

MarkRivers, 2019-10-21:
It looks like you must be running ADCore R3-3 or earlier because asynNDArrayDriver::decrementQueuedArrayCount() is calling callParamCallbacks(), which it no longer does in more recent releases.

Please look at the release notes for R3-4 and R3-5.  There was a problem with mutex locking when handing queuedArrayCount, which sounds like exactly what you are seeing.  Please update to at least R3-5 and see if that fixes the problem.

MarkRivers, 2019-10-22:
Have you had a chance to test this?

pheest, 2019-10-22:
Hi Mark,

I updated to newest releases (yesterday):
ADCore 3.3.2 to 3.7 and
ADSupport 1.5 to 1.9.

Then I saw your mail about new releases, 3.8 et. al.
Sometimes it's a little hard to keep up...!

I still experience a problem, but it is a different problem.
I'm gettin' a debug assertion in NDFileHDF5LayoutAML.cpp in LayoutXML::new_group,  at xmlFree(group_name).

I'm still assessing this.

MarkRivers, 2019-10-22:
There are 2 commits to look at:

```
commit 8747518ebd6d011a12158cb4b81f238a604ee79d
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Wed Mar 20 18:23:15 2019 -0500

    Added missing xmlFree calls for each string allocated by xmlTextReaderGetAttribute()

commit 6bdb0e9ebdaee365672bb048dd94b3f51f37a261
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Wed Nov 28 17:13:28 2018 -0600

    Define xmlFree to be free on Visual Studio; eliminates linker warnings
```
The xmlFree calls were added when were tracking down memory leaks.  The definition of xmlFree to be free was done to eliminate compiler warnings.  It works fine on release builds with Visual Studio, but perhaps there is an issue with debug builds?

I am closing this issue here, since it is not an asyn issue, it is an ADCore issue.  Please create a new issue in ADCore if it seems there is indeed a problem.

closed 2019-10-22 by MarkRivers