epics-modules/asyn #95: CentOS 8 build fails due to removal of rpc

jwlodek, 2019-11-08:
It seems that as of Fedora 28, `rpc` is being removed from the `glibc-headers`, which causes building `asyn` to fail:

https://fedoraproject.org/wiki/Changes/SunRPCRemoval

I am building asyn in a fresh docker container created with the following:
```

FROM centos:8
RUN dnf check-update || { rc=$?; [ "$rc" -eq 100 ] && exit 0; exit "$rc"; }
RUN dnf -y update
RUN dnf -y install epel-release
RUN dnf -y install git wget tar make cmake gcc gcc-c++ pkgconfig
RUN dnf -y install libtirpc-devel
RUN dnf -y --enablerepo=PowerTools install re2c libusb-devel rpcgen
RUN dnf -y install libxml2-devel pcre-devel libXext-devel
RUN dnf -y install libjpeg-devel readline-devel libusbx-devel
RUN dnf -y install boost-devel
RUN dnf -y install libraw1394
CMD ["/bin/bash"]
```
and 
```
docker build -t cos8test .
```
and
```
docker run -it cos8test
```
And I get the following error when running `make -sj` in the `asyn` directory:
```
In file included from vxi11core_xdr.c:6:
vxi11core.h:9:10: fatal error: rpc/rpc.h: No such file or directory
 #include <rpc/rpc.h>
          ^~~~~~~~~~~
compilation terminated.
../../asyn/asynRecord/asynRecord.c: In function ‘getOptions’:
../../asyn/asynRecord/asynRecord.c:1863:66: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(pasynRecPvt->old.hostinfo, pasynRec->hostinfo, sizeof(pasynRec->hostinfo));
                                                                  ^
../../asyn/asynRecord/asynRecord.c:1943:70: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
         strncpy(pasynRecPvt->old.hostinfo, pasynRec->hostinfo, sizeof(pasynRec->hostinfo));
                                                                      ^
make[3]: *** [/epics/base/configure/RULES_BUILD:234: vxi11core_xdr.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from ../../asyn/vxi11/drvVxi11.c:41:
../../asyn/vxi11/osiRpc.h:12:10: fatal error: rpc/rpc.h: No such file or directory
 #include <rpc/rpc.h>
          ^~~~~~~~~~~
compilation terminated.
make[3]: *** [/epics/base/configure/RULES_BUILD:234: drvVxi11.o] Error 1
```

MarkRivers, 2019-11-08:
If you look at asyn/configure/CONFIG_SITE you will see these lines:
```
# Some linux systems moved RPC related symbols to libtirpc
# To enable linking against this library, uncomment the following line
# TIRPC=YES
```
I think you just need to uncomment that last line.

jwlodek, 2019-11-12:
Looks like that's right, I uncommented that line, and it built just fine. Thanks!

closed 2019-11-12 by jwlodek

huyong1979, 2021-04-19:
Mark, I had the same problem on Red Hat 8.3. "TIRPC=YES" works. Just one suggestion: I am using RELEASE.local and I do not want modify CONFIG_SITE. I define "TIRPC=YES" in my RELEASE.local and it works. Could you move these lines to RELEASE? I usually create RELEASE.local from the existing RELEASE.
    # Some linux systems moved RPC related symbols to libtirpc
    # To enable linking against this library, uncomment the following line
    # TIRPC=YES

jwlodek, 2021-04-19:
I think that generally the RELEASE.* files only point at modules and paths to external dependencies, while to CONFIG_SITE.* files specify building options like this. Maybe a CONFIG_SITE.local makes sense?

MarkRivers, 2021-04-19:
asyn/configure/CONFIG_SITE already includes asyn/configure/CONFIG_SITE.local if it exists.  That is where you should put TIRPC=YES if you don't want to modify CONFIG_SITE.

In my case I build multiple versions of Linux in the same tree using different EPICS_HOST_ARCH, so I use the file CONFIG_SITE.linux-x86_64-centos8.Common.  That only uses TIRPC=YES on Centos8, not on Centos7.
```
corvette:~/devel/asyn/configure>more CONFIG_SITE.linux-x86_64-centos8.Common
TIRPC=YES
```

huyong1979, 2021-04-20:
Thank, Jakub and Mark. CONFIG_SITE.local works.

mdavidsaver, 2022-03-23:
A note for anyone else who missed it.  The `rpcgen` executable moved to a separate `rpcgen` RPM in the PowerTools repo.

mdavidsaver, 2022-03-23:
A further gotcha.  It looks like DNF (circa 4.7) treats repository names as case sensitive.  So the example above (https://github.com/epics-modules/asyn/issues/95#issue-520106915) does not work as written.

```
# dnf -y --enablerepo=PowerTools install re2c libusb-devel rpcgen
Error: Unknown repo: 'PowerTools'
```

```
# dnf -y --enablerepo=powertools install re2c libusb-devel rpcgen
CentOS Stream 8 - PowerTools   ...
...
```

```
# dnf --version
4.7.0
  Installed: dnf-0:4.7.0-7.el8.noarch at Sun 20 Feb 2022 07:22:12 PM GMT
  Built    : CentOS Buildsys <bugs@centos.org> at Sat 05 Feb 2022 07:07:55 AM GMT

  Installed: rpm-0:4.14.3-21.el8.x86_64 at Sun 20 Feb 2022 06:49:55 PM GMT
  Built    : CentOS Buildsys <bugs@centos.org> at Tue 11 Jan 2022 05:06:54 AM GMT
```