epics-modules/asyn #99: "main" blocking for thread "asynPortDriverCallback" to exit

hinxx, 2020-01-15:
I have an asynPortDriver based asyn driver that reports messages below when IOC shell 'exit' is requested. The operation would hang (forever?) until Ctrl+C is pressed to actually exit to shell. This is using asyn master branch code at commit 2eb4218bac334a8fde5eb3a5e7a7f0c4d616a48e.

```
epics> exit
[INF] AKI2C_TCA9555::~AKI2C_TCA9555: shut down ..
[INF] AKI2C::~AKI2C: shut down ..
[INF] AKBase::~AKBase: shut down ..
epicsThread::~epicsThread(): "main" blocking for thread "asynPortDriverCallback" to exit
was epicsThread object destroyed before thread exit ?
^C
```

In the more recent asyn master commit 9f4bcbd409b5d856bc7d87a9e35f189d666eecf1 the output is a bit different, but still hangs (without changes to module code, just recompiled asyn is used now).

```
epics> exit
[INF] AKI2C_TCA9555::~AKI2C_TCA9555: shut down ..
[INF] AKI2C::~AKI2C: shut down ..
[INF] AKBase::~AKBase: shut down ..
epicsThread::~epicsThread(): "_main_" blocking for thread "!" to exit
was epicsThread object destroyed before thread exit ?
epicsThread::~epicsThread(): "_main_" blocking for thread "!" to exit
was epicsThread object destroyed before thread exit ?
^C
```

MarkRivers, 2020-01-15:
What version of base are you using?

What do your 3 destructors look like?

mark0n, 2020-01-15:
Also which OS are you using?

hinxx, 2020-01-15:
It is on Linux Mint 19.x.
Using EPICS base 7, master branch couple of days old at commit 468f965dc2753e7a381dcc09dd9b5656e3149ac2.

Destructors are as follows:
`AKI2C_TCA9555::~AKI2C_TCA9555()` is empty, except message print.
`AKI2C::~AKI2C()` is empty, except message print.
`AKBase::~AKBase()` contains this (along with the message print):
```
pasynOctetSyncIO->disconnect(mAsynUserCommand);
```

This is to disconnect the pasynOctetSyncIO made in constructor:
```
	status = pasynOctetSyncIO->connect(mIpPort, 0, &mAsynUserCommand, NULL);
```

mark0n, 2020-01-15:
Your destructors look reasonable. [We're actually doing something very similar](https://github.com/mark0n/drvFGPDB/blob/release/fc3/drvFGPDBApp/src/drvFGPDB.cpp#L143) but we don't see these messages (Base 3.15.7 and Asyn 4.37).

How long has your IOC been running before you shut it down? The `asynPortDriverCallback` thread is supposed to shut down shortly after IOC start. If you run the IOC in the debugger you should see a message when the thread shuts down. I'm wondering if the thread has completed when you're trying to shut down the IOC. Or maybe the `asynPortDriverCallback` thread is getting stuck? Maybe due to a locking issue?

I'm surprised that the thread name changed between the two versions - "!" doesn't look like thread name to me. Could you please check if 1b3bdb5fa7b3f50049bf9586c5689996f03b8a2c is the commit that changes behaviour? Can you rule out memory corruption (e.g. by running with valgrind or thread sanitizer)?

MarkRivers, 2020-01-15:
In this commit 1b3bdb5fa7b3f50049bf9586c5689996f03b8a2c on January 11 I moved the call to thread.exitWait() from the destructor to the end of the run() method to avoid a warning message on vxWorks.  

What happens if you change the code to revert that commit?

anjohnson, 2020-01-15:
The thread named "!" might be some kind of a use-after-free if a pointer to the thread's name was requested while it was still alive but used after it has exited and cleaned. Mark's commit 1b3bdb5 on January 11 changed when the thread cleanup happens and will probably have resulted in the thread name being passed to free() as soon as the run() method returns, immediately invalidating the name pointer. To continue to do that you'd need to `epicsStrDup()` the name (and remember to `free()` it later).

However since the hang behavior itself hasn't changed I don't think that's what is causing the original problem.

MarkRivers, 2020-01-15:
How is your first destructor being invoked?

In areaDetector and other modules we typically are creating objects on the heap, and their destructors are thus never being called unless an epicsAtExit handler explicitly does so.  So we may not have noticed this problem.

hinxx, 2020-01-16:
@mark0n here is how the thread situation looks like right after startup. Assuming `asynPortDriverCa` is the offending thread, it exists right after startup, and it disappears after checking a couple of seconds later.

```
iocRun: All initialization complete
###############################################################################
create_monitor_set("auto_settings.req",30,"P=LAB:RFSW-001:,R=")
date
2020/01/16 08:43:37.139861
###############################################################################
epicsThreadShowAll
            NAME       EPICS ID   LWP ID   OSIPRI  OSSPRI  STATE
          _main_      0x1bf2200    24442      0       0       OK
          errlog      0x1bfbc90    24444     10       0       OK
      timerQueue      0x1c06180    24446     60       0       OK
        I2C_COMM      0x1c07d60    24448     50       0       OK
      timerQueue      0x1c06a50    24447     60       0       OK
          taskwd 0x7f43ac000af0    24449     10       0       OK
       TCA9555.1      0x1c0c680    24450     50       0       OK
asynPortDriverCa      0x1c0d3a0    24451     50       0       OK
      timerQueue      0x1c10960    24452     70       0       OK
           cbLow      0x1c14dd0    24453     59       0       OK
        cbMedium      0x1c35550    24454     64       0       OK
          cbHigh      0x1c35840    24455     71       0       OK
        dbCaLink      0x1c35c50    24456     50       0       OK
        scanOnce      0x1c96eb0    24457     67       0       OK
         scan-10      0x1ca34e0    24458     60       0       OK
          scan-5      0x1ca3730    24459     61       0       OK
          scan-2      0x1ca3980    24460     62       0       OK
          scan-1      0x1ca3bd0    24461     63       0       OK
        scan-0.5      0x1ca3e20    24462     64       0       OK
        scan-0.2      0x1ca4070    24463     65       0       OK
        scan-0.1      0x1ca42c0    24464     66       0       OK
         CAS-TCP      0x1cacc70    24465     18       0       OK
         CAS-UDP      0x1cacec0    24466     16       0       OK
      CAS-beacon      0x1cad110    24467     17       0       OK
    save_restore      0x1cad720    24468     20       0       OK
epics> auto_settings.sav: 0 of 0 PV's connected

epics> 
epics> date
2020/01/16 08:43:45.684794
epics> epicsThreadShowAll
            NAME       EPICS ID   LWP ID   OSIPRI  OSSPRI  STATE
          _main_      0x1bf2200    24442      0       0       OK
          errlog      0x1bfbc90    24444     10       0       OK
      timerQueue      0x1c06180    24446     60       0       OK
        I2C_COMM      0x1c07d60    24448     50       0       OK
      timerQueue      0x1c06a50    24447     60       0       OK
          taskwd 0x7f43ac000af0    24449     10       0       OK
       TCA9555.1      0x1c0c680    24450     50       0       OK
      timerQueue      0x1c10960    24452     70       0       OK
           cbLow      0x1c14dd0    24453     59       0       OK
        cbMedium      0x1c35550    24454     64       0       OK
          cbHigh      0x1c35840    24455     71       0       OK
        dbCaLink      0x1c35c50    24456     50       0       OK
        scanOnce      0x1c96eb0    24457     67       0       OK
         scan-10      0x1ca34e0    24458     60       0       OK
          scan-5      0x1ca3730    24459     61       0       OK
          scan-2      0x1ca3980    24460     62       0       OK
          scan-1      0x1ca3bd0    24461     63       0       OK
        scan-0.5      0x1ca3e20    24462     64       0       OK
        scan-0.2      0x1ca4070    24463     65       0       OK
        scan-0.1      0x1ca42c0    24464     66       0       OK
         CAS-TCP      0x1cacc70    24465     18       0       OK
         CAS-UDP      0x1cacec0    24466     16       0       OK
      CAS-beacon      0x1cad110    24467     17       0       OK
    save_restore      0x1cad720    24468     20       0       OK
epics> 
epics> exit
[INF] AKI2C_TCA9555::~AKI2C_TCA9555: shut down ..
[... 71 lines cut ...]
 callbackThread::~callbackThread()
 {
     shutdown.signal();
+    thread.exitWait();
 }
 
 /* I thought this would be a temporary fix until EPICS supported PINI after interruptAccept, which would then be used
@@ -933,7 +934,7 @@ void callbackThread::run()
         pPortDriver->callParamCallbacks(addr, addr);
     }
     epicsMutexUnlock(pPortDriver->mutexId);
-    thread.exitWait();
+    //thread.exitWait();
 }
```

@MarkRivers I do have an exit handler registered using `epicsAtExit` in AKI2C_TCA9555 constructor:
```
static void exitHandler(void *drvPvt) {
	AKI2C_TCA9555 *pPvt = (AKI2C_TCA9555 *)drvPvt;
	delete pPvt;
}
```

Coming back to the revert of the asyn code above. I'm confused now because in the report I was initially working with an older asyn code and that did not have the commit I reverted, but still resulted in the exit command hanging. I'll recompile all everything from scratch for both asyn commits to double check!

hinxx, 2020-01-16:
I've just recompiled the epics-base and all modules + IOC for for the two cases of asyn again. Epics base is at commit 468f965dc2753e7a381dcc09dd9b5656e3149ac2.

First case with asyn commit 2eb4218bac334a8fde5eb3a5e7a7f0c4d616a48e **exits OK**. This is different from what I have initially reported. I probably must have not ran the `make distclean` on epics base and all the modules of the IOC before `make all`. Sorry for false reporting an issue on that commit.

For the second case, using asyn commit 9f4bcbd409b5d856bc7d87a9e35f189d666eecf1 I do see the issue of exit hanging as initially reported.

That being said, I can understand why reverting of the asyn commit https://github.com/epics-modules/asyn/commit/1b3bdb5fa7b3f50049bf9586c5689996f03b8a2c solves the issue for me (as reported in the previous comment) - double checked again.

hinxx, 2020-01-16:
FWIW, I sprinkled some printf()'s in the epics-base code. I might over my head here, and can not make any judgement calls but here is the observation.

This is using asyn https://github.com/epics-modules/asyn/commit/9f4bcbd409b5d856bc7d87a9e35f189d666eecf1.

It seems the hang is at https://github.com/epics-base/epics-base/blob/468f965dc2753e7a381dcc09dd9b5656e3149ac2/modules/libcom/src/osi/epicsThread.cpp#L171. The `this->terminated` is false at this point and never gets true while waiting.

As expected, the thread does get out of the asyn `callbackThread::run()` but the https://github.com/epics-base/epics-base/blob/468f965dc2753e7a381dcc09dd9b5656e3149ac2/modules/libcom/src/osi/epicsThread.cpp#L115 is not entered at this point ; `threadDestroyed` is true hence the `this->terminated` is not set to true. `threadDestroyed` was set to true with the call to `thread.exitWait();` at the end of asyn `callbackThread::run()`.

Upon `exit` in iocsh, a call to `this->exitWait ( 10.0 )` from `epicsThread::~epicsThread ()` invoked by the asyn `callbackThread::~callbackThread()` does not enter https://github.com/epics-base/epics-base/blob/468f965dc2753e7a381dcc09dd9b5656e3149ac2/modules/libcom/src/osi/epicsThread.cpp#L152 and hits the `this->terminated` being false where it spins..

I commented out the asyn call to `thread.exitWait();` at the end of asyn `callbackThread::run()`. Now the https://github.com/epics-base/epics-base/blob/468f965dc2753e7a381dcc09dd9b5656e3149ac2/modules/libcom/src/osi/epicsThread.cpp#L115 is entered after the asyn `callbackThread::run()` finishes, and sets `this->terminated` to true.. When the `this->exitWait ( 10.0 )` from `epicsThread::~epicsThread ()` is executed is sees `this->terminated` being set to true and does not spin.
AFAICT, this commenting out call to `thread.exitWait();` is pretty much the same change as the revert of the code done previously.

MarkRivers, 2020-01-16:
Commit 1b3bdb5fa7b3f50049bf9586c5689996f03b8a2c was done on suggestion by @anjohnson to eliminate a warning at iocInit on vxWorks. This is discussed in this core-talk thread:  https://epics.anl.gov/core-talk/2020/msg00034.php.

@anjohnson is there a way to eliminate the warning without causing the hanging when the IOC exits?

anjohnson, 2020-01-23:
So after writing a series of tests for the epicsThread C++ API I can now answer this.

There is a bug in the implementation of the epicsThread class which makes calling `exitWait()` the wrong thing to do if the code *does* later delete the epicsThread object, since that destructor waits for the thread's internal `epicsEvent exitEvent` that never gets triggered in this case. After 10 seconds the destructor outputs the messages that @hinxx reported. This bug has been present for a very long time and I have a fix for it, but all the internal uses of the epicsThread class in Base call `exitWait()` from their parent thread so the code in Base has never hit this problem. I have a fix for it which I will apply, but that doesn't help Asyn today — sorry for originally suggesting this.

The internal design of the epicsThread class allows a thread's `run()` method to delete its own epicsThread object before it returns. There are comments and very carefully designed code included in the class that explicitly allows this, and the result does work as long as the parent thread never tries to use the epicsThread object after it might have been destroyed. Jeff Hill created several classes in Base that do `delete this;` in this kind of circumstance, but while that does work if you're careful, I think it likely to be harder for maintainers to understand and thus to maintain. I also worry that some compiler optimizer might do something that breaks this in the future.

Thus my suggested solution is to change the `callbackThread::thread` member into a _pointer to_ an epicsThread; personally I would rename it to `pthread` at the same time, and replace all uses of `thread.` with `pthread->`. You can still create it in the `callbackThread` constructor (but now using `pthread(new epicsThread(*this, ...))`), but you must explicitly delete it in the `callbackThread::run()` method. I would also set `pthread = NULL;` after the `delete pthread;` just in case.

MarkRivers, 2020-01-24:
I have implemented that suggestion by @anjohnson and pushed to the master branch.

@hinxx can you please test to see if it fixes your problem with the destructors?

mark0n, 2020-01-24:
Some of our drivers including [drvFGPDB](https://github.com/mark0n/drvFGPDB) are segfaulting if I compile them against `master` (8b9210268602440d7e40b1df57ee79b6a3171dc2) and run their tests. After reverting ea7be02ec7a528ff24fbdfa92b25d3baa805abf5 and 1b3bdb5fa7b3f50049bf9586c5689996f03b8a2c all our tests pass.

I'm not sure if calling `delete pthread` in the `callbackThread::run()` is a good idea. Wouldn't this delete a thread object that is still executing the `run()` function? Isn't destroying an un-joined thread undefined behaviour?

MarkRivers, 2020-01-24:
I think that reverting both of those gets us back to the original problem, which was warning messages on vxWorks shortly after iocInit.

mark0n, 2020-01-24:
That's right. The problem is that the current solution doesn't work either... I would expect something like the following:
```
callbackThread::~callbackThread()
{
    shutdown.signal();
    pthread->join();
    delete pthread;
    pthread = NULL;
}

void callbackThread::run()
{
    int addr;
    while(!interruptAccept && !shutdown.tryWait())
        epicsThreadSleep(0.001);
    epicsMutexLock(pPortDriver->mutexId);
    for (addr=0; addr<pPortDriver->maxAddr; addr++) {
        if(shutdown.tryWait()) break;
        pPortDriver->callParamCallbacks(addr, addr);
    }
    epicsMutexUnlock(pPortDriver->mutexId);
}

```

MarkRivers, 2020-01-24:
What about adding 
```delete this;```
at the end of run().  Then the destructor is run and thread isn't handing around after it is done

anjohnson, 2020-01-24:
@mark0n Mark's and Hinko's use-cases for callbackThread are trying to _not_ join with the thread. Mark needs to tell his epicsThread that it will _never_ be joined, and that it should clean up itself when it's finished doing the work it was created for. It sounds like your use-case is different and that you may need a different model, but I don't know exactly where this thread is used since I haven't developed anything using Asyn recently. Can you point out where in your code you use callbackThread? A back-trace from the segfault might be helpful too.

The epicsThread C++ API was designed (by Jeff Hill) to allow applications to delete an epicsThread object from its run() method. Take a look at the [epicsThreadClassTest.cpp](https://github.com/epics-base/epics-base/commit/dbd6f7e8079bb158b9b30eaec6a60d0327752d6a) program that I added to 3.15 yesterday for proof — I have run these tests against the latest 7.0 but haven't merged up from 3.15 yet since I also have a fix to epicsThread to add. Mark has implemented the "BdrS" test case in Asyn, following my recommendation.

anjohnson, 2020-01-24:
Note that Mark can't use pthread->join() since that method was only added in the 7.0 branch and Asyn must build against much older versions of Base.

mark0n, 2020-01-24:
We need to wait for the thread function to complete before we return from `callbackThread::~callbackThread()`. This is the only way I can think of to ensure `callbackThread::run()` has returned before the portDriver object gets destroyed. Destroying the portDriver while the thread is still running would cause havoc (e.g. when using `pPortDriver->mutexId`). And, yes, this is relevant in practice: Our tests are running pretty quickly so they are calling the destructor faster than the thread gets spawned and completes.

I wish we could simply use `std::thread` which comes with `join()` - but once again that requires C++11 :disappointed:

BTW: [Some advise on how to use `delete this`](https://isocpp.org/wiki/faq/freestore-mgmt#delete-this). I still hope we can avoid it, though.

anjohnson, 2020-01-24:
@mark0n You don't actually need a `join()` then, since you don't care if the child thread finishes and cleans up before for the parent is ready to run the destructor. The problem with `join()` for Mark's use-case is that the child thread stays alive and doesn't release its memory until the parent actually performs the cleanup, and for his VxWorks IOCs that never happens so he's wasting resources.

To implement your requirement, the callbackThread object just needs an `epicsEvent doneEvt` member and the `run()` method calls `doneEvt.signal()` just before it returns. The callbackThread destructor calls `doneEvt.wait()`. This doesn't stop the child thread from exiting and cleaning up using the code that Mark has already committed, but it does delay the parent inside the destructor until that has happened.

There is no need for `delete this;` (which you will find scattered around Jeff Hill's code in Base) and the result is compatible back to Base 3.14.x.

MarkRivers, 2020-01-24:
I think I understood what Andrew is suggesting. I have committed that, and the asynPortDriver unittests run OK.

@mark0n and @hinxx please test as well.

mark0n, 2020-01-27:
Thanks for fixing this issue. I have tested 6e1832014447a419a555fbb436d1c009dfa5b727. It works well (all our tests pass, Valgrind doesn't report any new issues).

hinxx, 2020-01-28:
This fixes it for my IOC as well. Thank you @MarkRivers !

closed 2020-02-24 by MarkRivers