epics-modules/asyn #108: support SO_REUSEPORT socket option

prjemian, 2020-04-18:
The [`SO_REUSEPORT`](https://lwn.net/Articles/542629/) socket option allows multiple clients on the same user@host to bind to the same **UDP** port.  With this option enabled in *asyn*, I expect to be able to run multiple IOCs using [new support](https://github.com/epics-modules/ip/blob/969799373765c57a4992bc33f5e7e227b1f453bc/documentation/Oxford_CS800.md) for the Oxford CS800 CryoStream controller (blows cold nitrogen gas at protein sample during crystallography measurement).  Currently, when I try to run a second IOC from the same user@host, this error is reported to the console and no communications is the result:

```
# ### cs800.iocsh ###
# STATUS_ADDR  "255.255.255.255:30304:30304 UDP"
# COMMAND_ADDR "10.0.0.173:30305 UDP*"
# Status Packets
drvAsynIPPortConfigure("OC_SP", "192.168.144.169:30304:30304 UDP*", 0, 0, 0)
2020/04/18 11:16:04.544 OC_SP -1 autoConnect could not connect: unable to bind to local port: Address already in use
# Commands
```

Is it possible to call `setsockopt()` ([similar to what is shown](https://lwn.net/Articles/542629/)) from the `st.cmd` file?

Otherwise, can support for the `SO_REUSEPORT` socket option be enabled in asyn?  The text `SO_REUSEPORT` does not appear anywhere in the asyn repo so I assume this option is not available yet.

prjemian, 2020-04-18:
Without this option, I must run each instance of this support on a separate workstation (or using a separate NIC if multiple NICs are available on the same workstation).

prjemian, 2020-04-18:
[Stackoverflow](https://stackoverflow.com/questions/14388706/how-do-so-reuseaddr-and-so-reuseport-differ?rq=1) has some entries about `SO_REUSEPORT`.

My goal is to run multiple IOCs on the same host from same username, each supporting a different CS800 controller.  Advice is welcome.

norumwe12, 2020-04-18:
I don’t see any major down side to adding this on systems which provide SO_REUSEPORT when an explicit local port is specified.  Just add the setsockopt() call after the bina() and inside an #ifdef SO_REUSEPORT conditional.  Maybe also with a warning message if the call fails.

I does make things yet a little more unportable, though — I’m not sure that I’d want to try to add all the required conditionals and code to support Windows, for example.

ralphlange, 2020-04-18:
I don't think this is going to work.

If the devices are sending UDP data to the same port on the IOC host (as the protocol file suggests), only the last IOC that was started will get all data. (Unless the devices are sending broadcasts. Then all IOCs will get all data. That's technically the same issue as with Channel Access name resolution not working properly on a Linux box running multiple IOCs if the client uses unicasts.)

Have a look at how to configure virtual network interfaces on your host - which is actually pretty simple on Linux. Then every IOC can use its own IP address for the device connection and all is good.

prjemian, 2020-04-18:
+1 for virtual NICs, thanks.

The CS800 controllers *broadcast* status once a second by UDP to port 30304.  Another problem to resolve is *all* status broadcasts being received by every IOC.  That seems to be be resolved by specifying the IP number of the controller:

```
epicsEnvSet("CS800_IP_ADDR", "10.0.0.173")
epicsEnvSet("STATUS_ADDR", "$(CS800_IP_ADDR):30304:30304 UDP")
epicsEnvSet("COMMAND_ADDR", "$(CS800_IP_ADDR):30305 UDP")
iocshLoad("$(IP)ipApp/Db/Oxford_CS800.iocsh", "PREFIX=$(PREFIX),INSTANCE=CS")
```

At least it works when I start an IOC and *two* simulated CS800 controllers on the same subnet.  I can test with *real* controllers on Monday when the beam is down.

prjemian, 2020-04-18:
The `iocshLoad` calls this [code](https://github.com/epics-modules/ip/blob/e849c5827e9eec646ee4ad1af9ea161ba5fc02f3/iocsh/Oxford_CS800.iocsh#L12-L18):

```
# STATUS_ADDR  "255.255.255.255:30304:30304 UDP"
# COMMAND_ADDR "10.0.0.173:30305 UDP*"

# Status Packets
drvAsynIPPortConfigure("OC_SP", "$(STATUS_ADDR)", 0, 0, 0)
# Commands
drvAsynIPPortConfigure("OC_CMD", "$(COMMAND_ADDR)", 0, 0, 0)
```

ralphlange, 2020-04-18:
Then a reused port is the same as multiple virtual NICs.
Can't you just run them all from one IOC? Doesn't look like too much traffic, and you probably don't have hundreds of them, either.

prjemian, 2020-04-18:
Trying it but it fails.

```
# ### cs800.iocsh ###
# STATUS_ADDR  "255.255.255.255:30304:30304 UDP"
# COMMAND_ADDR "10.0.0.173:30305 UDP*"
# Status Packets
drvAsynIPPortConfigure("OC_SP", "192.168.144.144:30304:30304 UDP", 0, 0, 0)
# Commands
drvAsynIPPortConfigure("OC_CMD", "192.168.144.144:30305 UDP", 0, 0, 0)
#asynSetTraceIOMask("OC_SP", 0, 2)
#asynSetTraceMask("OC_SP", 0, 9)
#asynSetTraceIOMask("OC_CMD", 0, 2)
#asynSetTraceMask("OC_CMD", 0, 9)
dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-R4-37/db/asynRecord.db","P=cs800:,R=CS:ASYN:SP,PORT=OC_SP,ADDR=0,OMAX=1000,IMAX=1000")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-R4-37/db/asynRecord.db","P=cs800:,R=CS:ASYN:CMD,PORT=OC_CMD,ADDR=0,OMAX=1000,IMAX=1000")
epicsEnvSet("STREAM_PROTOCOL_PATH", ".:/usr/local/epics/synApps_6_1/support/ip-git/ipApp/Db")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/ip-git/db/Oxford_CS800.db", "P=cs800:,Q=CS,PORT=OC_SP,PORTCMD=OC_CMD")
drvAsynIPPortConfigure("OC_SP2", "192.168.144.169:30304:30304 UDP", 0, 0, 0)
2020/04/18 14:54:59.262 OC_SP2 -1 autoConnect could not connect: unable to bind to local port: Address already in use
drvAsynIPPortConfigure("OC_CMD2", "192.168.144.169:30305 UDP", 0, 0, 0)
```

Changing `:30304:30304` to `:30304` starts with no errors.  But no PV updates (from streams support).

prjemian, 2020-04-18:
Summary is that any and all status broadcasts will come in to asyn configured:

```
drvAsynIPPortConfigure("OC_SP", "255.255.255.255:30304:30304 UDP", 0, 0, 0)
```

which moves the problem a little further along to stream: how to separate the status broadcasts by their sender?  The IP and MAC addresses of the sender are in the packet header and the IP address is also in the packet data.  A C or SNL program could filter out the status from the desired controller but filtering by IP sender that is probably beyond the capability of a stream protocol.

ralphlange, 2020-04-18:
Ok. When you want to filter by sender address by configuring UDP "connections", you need a separate for each - then we need the `SO_REUSEPORT` again.
Trying to have Stream split the messages apart? Sounds complex. Probably needs a separate protocol for each device so that Stream can match the right one. Or some database plumbing where Stream puts the data and the IP address in shared records and fires a bunch of calcout/seq combinations that forward the data to the right target records if the IP matches.

`SO_REUSEPORT` or virtual NICs sound like a more obvious approach.

prjemian, 2020-04-18:
I agree.  How hard is it to add?

prjemian, 2020-04-18:
And for now, support each CS800 controller from a separate host.

ralphlange, 2020-04-18:
virtual NICs should work for now.

prjemian, 2020-04-18:
That's much easier on the home LAN than the working beam line.  They are *very* risk averse since they can do virus-related work now.

prjemian, 2020-04-18:
Thanks for all the help!

ralphlange, 2020-04-18:
As Eric mentioned: the change itself is not too hard to do. A single call on the socket. Windows portability has been solved for the Channel Access case (in the UDP server).

Question is if Mark wants to keep the sockets single-use by default (which might make sense) and - if so - which would be the appropriate way to configure the REUSEPORT option.

Welcome!

ralphlange, 2020-04-19:
One message from Eric does not show up here - I just got the mail, don't know what happened:

> The driver does a connect() so I would think that packets from the hosts in question would be delivered to the correct sockets even if the local address/port is shared.  As far as I understand things, a socket is associated with 5 attributes, namely a protocol, source address, source port, destination address, destination port, so the network stack has enough information for packets to be delivered to the right socket.
> e.g.
> 	Socket	Protocol	Dest Addr	Dest Port		Source Addr			Source Port
> 	  1		UDP        Host A              Port A           	My IP address			Shared Port
> 	  2		UDP	Host B		Port B		My IP address			Shared Port
> 
> So incoming packets from Host A, Port A would be delivered to socket 1 and those from Host B, Port B, would be delivered to socket 2, right?
> Or is this one of those corner cases that a significant number (all?) kernels get wrong?

ralphlange, 2020-04-19:
Well, I always thought that `connect()` on a UDP socket only stores the default peer data, so that the application can use `send()` and `recv()` instead of the fully specified `_from()` versions of the calls. (And be more efficient, as routing does not have to be checked for every send/recv call.)

I do agree that the kernel _should have_ all necessary information to get the routing right. I never worked with such a UDP "connection", so I can't say if this setup falls under the UDP server case where all non-broadcast packages get delivered to exactly one process waiting on the shared port.

Nevertheless: I don't think adding the `SO_REUSEPORT` specifically for the local port of a UDP client "connection" setup would change behavior that existing applications rely on. (So, no need to introduce an option to the user visible API.)

MarkRivers, 2020-04-19:
Questions:
- Does SO_REUSEPORT only make sense for UDP, or might it be useful for TCP as well?
- For UDP does it only make sense if broadcast is also enabled as in Pete's example?

I can think of 2 ways to handle the API.

1) Introduce a new character in the protocol that means SO_REUSEPORT, for example &.  Then depending on the answers to the above questions new protocols would be
  - udp&
  - udp*&
  - tcp&
2) Use the asynOption interface, like the following.
```
# Create the port with autoconnect=0.
drvAsynIPPortConfigure("OC_SP", "192.168.144.169:30304:30304 UDP*", 0, 1, 0)
# Enable SO_REUSEPORT
asynSetOption("OC_SP", 0,  "SO_REUSEPORT", "Y")
# Now enable autoconnect
asynAutoConnect("OC_SP", 0,  1)
```
I will work on this once I hear a consensus on which variations make sense.

MarkRivers, 2020-04-19:
Sorry, I just noticed Ralph's comment:
>Nevertheless: I don't think adding the SO_REUSEPORT specifically for the local port of a UDP client "connection" setup would change behavior that existing applications rely on. (So, no need to introduce an option to the user visible API.)

However, this article explains the uses of SO_REUSEPORT, and how it can be useful for both TCP and UDP.  

https://lwn.net/Articles/542629/

But for TCP I don't think we want to make it the default, so we do need an API change to handle it.

MarkRivers, 2020-04-19:
I have created a new branch of asyn, SO_REUSEPORT.  It implements option 1) above, i.e. new protocols:
- udp&
- udp*&
- tcp&

On _WIN32 I used the socket option SO_REUSEADDR because SO_REUSEPORT is not supported.  This article says that it does what Pete wants for UDP broadcast sockets:
https://stackoverflow.com/questions/13637121/so-reuseport-is-not-defined-on-windows-7

On all other architectures it uses the socket option SO_REUSEPORT.

I have tested that it compiles on Linux, vxWorks, and Windows VS2015.

@prjemian can you see if it works for you on Linux?

prjemian, 2020-04-19:
Thanks for the quick response.  I'll get to it later today.  Faster if the weather goes bad but that's not likely.  If this works, this can save me from a rewrite of the CS800 support in SNL.

norumwe12, 2020-04-19:
Right — here’s the salient section of the connect(2) man page — I infer from, “only address from datagrams are to be received”, that this should take care of Pete Jemian’s problem.

DESCRIPTION
     The parameter socket is a socket.  If it is of type SOCK_DGRAM, this call
     specifies the peer with which the socket is to be associated; this address
     is that to which datagrams are to be sent, and the only address from which
     datagrams are to be received.  If the socket is of type SOCK_STREAM, this
     call attempts to make a connection to another socket.  The other socket is
     specified by address, which is an address in the communications space of the
     socket.

> On Apr 19, 2020, at 1:36 AM, Ralph Lange <notifications@github.com> wrote:
> 
> Well, I always thought that connect() on a UDP socket only stores the default peer data, so that the application can use send() and recv() instead of the fully specified _from() versions of the calls. (And be more efficient, as routing does not have to be checked for every send/recv call.)

ralphlange, 2020-04-20:
Thanks a lot, Mark. The change looks good.

This will allow adding the SO_REUSEPORT flag for all client type configurations, so that the originating port of the connection can be shared. Server type ports are still exclusive. IMHO this is ok (as it would directly lead to the issue with incoming broadcasts vs. unicasts) and we can relax and wait for someone to actually need that feature. Just wanted to make it clear.

prjemian, 2020-04-20:
Working with two CS800 simulators:

PREFIX | IP | controller ID
--- | --- | ---
`cs800:` | 192.168.144.144 | 144
`cryo:` | 192.168.144.113 | 113

## `UDP*&`

Starting support for both from the same linux-x86_64 soft IOC (running on 192.168.144.99) using `UDP*&` for both status and command ports:

<details>
<summary>st.cmd configuration</summary>

```
# ----------------------------------------------------------
# >>> first CS800 ----------------
# Oxford CryoStream 800 cooler
epicsEnvSet("CS800_IP_ADDR", "192.168.144.144")
epicsEnvSet("STATUS_ADDR", "$(CS800_IP_ADDR):30304:30304 UDP*&")
epicsEnvSet("COMMAND_ADDR", "$(CS800_IP_ADDR):30305 UDP*&")
iocshLoad("$(IP)/iocsh/Oxford_CS800.iocsh", "PREFIX=$(PREFIX),INSTANCE=CS")
# <<< first CS800 ----------------
# ----------------------------------------------------------

# ----------------------------------------------------------
# >>> second CS800 ----------------
epicsEnvSet("CS800_IP_ADDR2", "192.168.144.113")
drvAsynIPPortConfigure("CS800_STATUS", "${CS800_IP_ADDR2}:30304:30304 UDP*&", 0, 0, 0)
drvAsynIPPortConfigure("CS800_COMMAND", "${CS800_IP_ADDR2}:30305 UDP*&", 0, 0, 0)

dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-git/db/asynRecord.db","P=cryo:,R=CS:ASYN:SP,PORT=CS800_STATUS,ADDR=0,OMAX=1000,IMAX=1000")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-git/db/asynRecord.db","P=cryo:,R=CS:ASYN:CMD,PORT=CS800_COMMAND,ADDR=0,OMAX=1000,IMAX=1000")
# epicsEnvSet("STREAM_PROTOCOL_PATH", ".:/usr/local/epics/synApps_6_1/support/ip-git/ipApp/Db")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/ip-git/db/Oxford_CS800.db", "P=cryo:,Q=CS,PORT=CS800_STATUS,PORTCMD=CS800_COMMAND")
# <<< second CS800 ----------------
# ----------------------------------------------------------
```

</details>

No errors on startup:

<details>
<summary>IOC console output on startup</summary>

```
# ----------------------------------------------------------
# >>> first CS800 ----------------
# Oxford CryoStream 800 cooler
epicsEnvSet("CS800_IP_ADDR", "192.168.144.144")
epicsEnvSet("STATUS_ADDR", "192.168.144.144:30304:30304 UDP*&")
epicsEnvSet("COMMAND_ADDR", "192.168.144.144:30305 UDP*&")
iocshLoad("/usr/local/epics/synApps_6_1/support/ip-git/iocsh/Oxford_CS800.iocsh", "PREFIX=cs800:,INSTANCE=CS")
# ### cs800.iocsh ###
# STATUS_ADDR  "255.255.255.255:30304:30304 UDP"
# COMMAND_ADDR "10.0.0.173:30305 UDP*"
# Status Packets
drvAsynIPPortConfigure("OC_SP", "192.168.144.144:30304:30304 UDP*&", 0, 0, 0)
# Commands
drvAsynIPPortConfigure("OC_CMD", "192.168.144.144:30305 UDP*&", 0, 0, 0)
#asynSetTraceIOMask("OC_SP", 0, 2)
#asynSetTraceMask("OC_SP", 0, 9)
#asynSetTraceIOMask("OC_CMD", 0, 2)
#asynSetTraceMask("OC_CMD", 0, 9)
dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-git/db/asynRecord.db","P=cs800:,R=CS:ASYN:SP,PORT=OC_SP,ADDR=0,OMAX=1000,IMAX=1000")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/asyn-git/db/asynRecord.db","P=cs800:,R=CS:ASYN:CMD,PORT=OC_CMD,ADDR=0,OMAX=1000,IMAX=1000")
epicsEnvSet("STREAM_PROTOCOL_PATH", ".:/usr/local/epics/synApps_6_1/support/ip-git/ipApp/Db")
dbLoadRecords("/usr/local/epics/synApps_6_1/support/ip-git/db/Oxford_CS800.db", "P=cs800:,Q=CS,PORT=OC_SP,PORTCMD=OC_CMD")
# <<< first CS800 ----------------
[... 31 lines cut ...]
cryo:CS:CTRLNUM                2020-04-20 11:52:36.563614 113
cryo:CS:CTRLNUM                2020-04-20 11:52:37.156794 144
cs800:CS:CTRLNUM               2020-04-20 11:52:37.156795 144
cryo:CS:CTRLNUM                2020-04-20 11:52:37.566597 113
cs800:CS:CTRLNUM               2020-04-20 11:52:37.566650 113
cs800:CS:CTRLNUM               2020-04-20 11:52:38.187451 144
cryo:CS:CTRLNUM                2020-04-20 11:52:38.187454 144
cryo:CS:CTRLNUM                2020-04-20 11:52:38.569569 113
cs800:CS:CTRLNUM               2020-04-20 11:52:38.569570 113
^C
```

</details>

Verified that command sent from EPICS PV to one controller is only received by the correct controller.  For both controllers. (this is correct)

## `UDP&`

Changing to `UDP&`  (remove the broadcast configuration) and same results for status and commands.  That is, each database receives all status updates (this is not correct).  Commands sent to intended controller (this is correct).

## `:30304:30304` vs. `:30304`

When this part is shortened, no status updates are received by either controller.

How can this be configured so that status updates from controller 144 are received only by the EPICS database for that controller?

norumwe12, 2020-04-20:
On Apr 20, 2020, at 10:49 AM, Pete R Jemian <notifications@github.com> wrote:
> 
> 
> Working with two CS800 simulators:
> 
> PREFIX	IP	controller ID
> cs800:	192.168.144.144	144
> cryo:	192.168.144.113	113
> UDP*&
> 
> Starting support for both from the same linux-x86_64 soft IOC (running on 192.168.144.99) using UDP*& for both status and command ports:
> 
> st.cmd configuration
> No errors on startup:
> 
> IOC console output on startup
> Watching for the controller ID for each instance. Both EPICS database instances receive status updates from both controllers (this is not correct).
> 
> Verified that command sent from EPICS PV to one controller is only received by the correct controller. For both controllers. (this is correct)
> 
> UDP&
> 
> Changing to UDP& (remove the broadcast configuration) and same results for status and commands. That is, each database receives all status updates (this is not correct). Commands sent to intended controller (this is correct)
> 
> 
This sounds like the network stack is not limiting reception to just those packets from the sender specified by the connect().  My interpretation of the connect(2) documentation is that the effect you’re seeing means that the network code is broken, but my interpretation doesn’t count for much!
> :30304:30304 vs. :30304
> 
> When this part is shortened, no status updates are received by either controller
> 

This is expected since the version without the local port specification (the second number above) just lets the kernel pick an unused port number.  The CS800 can’t handle this since it has the destination port wired in.
> How can this be configured so that status updates from controller 144 are received only by the EPICS database for that controller?
> 
> 
Fix the network stack?
Place the devices on different virtual NICs?
Tell the manufacturer to use a less bizarre communications method?

MarkRivers, 2020-04-20:
@prjemian I think the simplest configuration is to use UDP& on the status port and UDP on the command port.  I don't think there is a need for the broadcast flag on the status port.  The broadcast flag is to allow EPICS to *send* broadcasts, nothing special needs to be done to receive them.  The command port does not need the & flag, it is just a normal UDP port.

>> How can this be configured so that status updates from controller 144 are received only by the EPICS database for that controller?

> Fix the network stack?
> Place the devices on different virtual NICs?
> Tell the manufacturer to use a less bizarre communications method?

@prjemian can you send the asynTrace output on the status port with asynSetTraceMask=TRACE_ERROR|TRACEIO_DRIVER so we can see the message content?  I think you said in a previous message that the status message contains the IP address of the machine sending the message?  If so then I think I can add to @norumwe12's list above:
- Write an asynInterpose layer.  This will be configured with the IP address of the desired sender.  It will discard messages that are not from that IP address.  That way you will only receive messages from the desired IP address.

prjemian, 2020-04-20:
@norumwe12 said:

> Fix the network stack?

I'm developing on a linux mint OS with most recent updates for deployment on RHEL7 at the APS.  Might bear to test on the APS network before proceeding with this.

> Place the devices on different virtual NICs?

already responded to this

> Tell the manufacturer to use a less bizarre communications method?

possible

prjemian, 2020-04-20:
@MarkRivers : the IP address is in the UDP packet *header*.  Can the asynRecord access that content?

Here's a few packets with `asynSetTraceMask=TRACE_ERROR|TRACEIO_DRIVER`:

```
ioccs800> 2020/04/20 14:22:42.131 192.168.144.144:30304:30304 UDP& (from 192.168.144.113:55932) read 928

2020/04/20 14:22:42.888 192.168.144.144:30304:30304 UDP& (from 192.168.144.144:47912) read 928

2020/04/20 14:22:43.131 192.168.144.144:30304:30304 UDP& (from 192.168.144.113:55932) read 928

2020/04/20 14:22:43.898 192.168.144.144:30304:30304 UDP& (from 192.168.144.144:47912) read 928

2020/04/20 14:22:44.131 192.168.144.144:30304:30304 UDP& (from 192.168.144.113:55932) read 928
```

repeating that with `traceIOHex` ON:

<details>

```
ioccs800> 
ioccs800> 2020/04/20 14:23:27.235 192.168.144.144:30304:30304 UDP& (from 192.168.144.113:55932) read 928

aa ab 03 98 03 e8 01 df 03 e9 01 b4 03 ea 39 f3 03 eb 37 a0 
03 ec 04 4f 03 ed 02 18 03 ee 02 27 03 f2 01 de 03 f3 02 19 
03 f4 00 20 03 f5 00 20 03 f6 01 ce 03 f7 01 cb 03 f8 01 de 
03 f9 02 17 03 fa 00 22 03 fb 02 22 03 fc 01 bf 03 fd 01 8d 
03 fe 02 08 03 ff 01 9d 04 00 01 ba 04 01 01 e2 04 02 01 b5 
04 03 01 f6 04 04 00 71 04 05 0c a4 04 06 0d 15 04 07 01 cb 
04 08 01 f7 04 09 01 9c 04 0a 02 4e 04 0b 02 1f 04 0c 01 e7 
04 0d 01 fc 04 10 02 1a 04 11 01 fb 04 12 01 cf 04 13 01 f6 
04 14 02 0e 04 15 02 3c 04 16 01 d0 04 1a 27 10 04 1b 27 1d 
04 1c 01 f5 04 1d 00 03 04 1e 00 03 04 1f 01 68 04 20 27 10 
04 21 38 f2 04 22 3d c2 04 23 04 23 04 24 00 11 04 25 00 1d 
04 26 00 1e 04 27 00 20 04 28 02 0a 04 29 01 d4 04 2a 00 03 
04 2b 02 36 04 2c 01 bc 04 2d 00 24 04 2e 00 19 04 2f 01 f0 
04 30 01 ff 04 31 01 ce 04 38 01 e8 04 39 02 0b 04 3a 02 00 
04 3b 01 d8 04 3c 01 f7 04 3d 01 e7 04 3e 01 b0 04 3f 01 9f 
04 40 00 1c 04 41 00 1c 04 42 00 25 04 43 01 e4 04 44 02 06 
04 45 01 ea 04 46 02 18 04 47 02 06 04 48 01 e8 04 49 01 f3 
04 4a 01 f1 04 4c 02 3d 04 4d 00 3a 04 4e 01 a6 04 4f 00 21 
04 50 01 f9 04 51 01 fc 04 52 01 ee 04 53 01 fe 04 54 01 f0 
04 b0 01 c7 04 b1 01 da 04 b2 01 fe 04 b3 00 1b 04 b4 02 13 
04 b5 02 41 04 b6 01 da 04 b7 02 59 04 b8 02 51 04 b9 01 f2 
04 ba 01 f9 04 bb 01 fd 04 bc 02 06 05 14 02 0d 05 15 01 a7 
05 16 01 f7 05 17 01 9a 05 18 01 ed 05 19 01 f6 05 1a 02 1e 
05 1b 02 03 05 1c 01 f7 05 1d 02 18 05 1e 02 35 05 1f 02 1b 
05 20 01 d6 05 21 02 07 05 22 01 da 05 78 01 9c 05 79 02 2c 
05 7a 01 f0 05 7b 01 db 05 7c 02 12 05 7d 01 b6 05 7e 01 b6 
05 7f 01 e0 05 80 02 3a 05 81 01 fc 05 82 01 d1 05 83 02 08 
05 84 01 f1 05 85 02 32 05 86 02 05 05 87 01 c9 05 88 02 05 
05 89 01 ee 05 8a 02 4e 05 8b 01 fb 05 8c 01 c5 05 8d 02 3a 
05 8e 01 db 05 8f 01 d0 05 90 01 f0 05 91 01 fa 05 92 01 9d 
05 93 01 c8 05 94 01 fe 05 95 01 cb 05 96 01 e5 05 97 01 c8 
05 98 01 fb 05 99 01 eb 05 dc 02 0c 05 dd 02 0f 05 de 02 38 
05 df 02 09 05 e0 01 e3 05 e1 01 fe 05 e2 01 b4 05 e3 02 00 
05 e4 02 38 05 e5 01 e8 05 e6 01 d1 05 e7 02 08 05 e8 02 20 
05 e9 01 e6 05 ea 01 e0 05 eb 01 ba 06 40 02 23 06 41 02 49 
06 42 02 0e 06 43 01 f9 06 44 02 06 06 45 01 ec 06 46 01 e4 
06 a4 01 da 06 a5 02 01 07 08 01 f7 07 09 01 f8 07 0a 01 91 
07 0b 01 e5 07 0c 02 12 07 0d 01 e6 07 0e 02 2c 07 0f 01 fe 
07 10 02 23 07 11 01 da 07 12 01 f3 07 13 02 2e 07 14 01 a3 
07 15 02 46 07 6c 01 b2 07 6d 01 d7 07 6e 01 9d 07 6f 01 ae 
07 d0 01 c6 07 d1 01 dc 07 d2 01 f4 07 da 3c 99 07 db 3c 5d 
07 dc 02 01 07 dd 3d db 07 de 00 22 07 df 00 22 07 e0 00 1a 
07 e1 00 15 07 e2 01 9b 07 e3 00 1e 07 e4 00 1f 07 e5 02 10 
07 e6 01 f4 07 e7 01 be 07 e8 01 e5 07 ee 39 87 07 ef 39 73 
07 f0 02 15 07 f1 01 9e 07 f2 01 f6 07 f3 37 f6 07 f4 3c dc 
07 f5 01 c3 07 f6 01 b9 07 f7 01 d8 07 f8 02 01 07 f9 3b 7a 
[... 132 lines cut ...]
05 98 01 f2 05 99 02 00 05 dc 01 af 05 dd 02 20 05 de 02 17 
05 df 01 f4 05 e0 02 18 05 e1 02 11 05 e2 01 f4 05 e3 01 f2 
05 e4 02 03 05 e5 01 fd 05 e6 01 74 05 e7 01 c3 05 e8 01 c4 
05 e9 01 e3 05 ea 02 08 05 eb 01 fb 06 40 02 22 06 41 01 c5 
06 42 02 3e 06 43 01 ff 06 44 02 20 06 45 01 f5 06 46 02 02 
06 a4 02 04 06 a5 01 ef 07 08 02 03 07 09 02 20 07 0a 01 f9 
07 0b 02 2a 07 0c 02 4c 07 0d 02 2b 07 0e 02 18 07 0f 01 c9 
07 10 01 ea 07 11 02 20 07 12 01 92 07 13 02 00 07 14 02 0e 
07 15 01 ce 07 6c 02 09 07 6d 02 04 07 6e 01 c1 07 6f 02 45 
07 d0 01 cf 07 d1 02 06 07 d2 01 db 07 da 39 d0 07 db 3a 29 
07 dc 01 db 07 dd 35 b1 07 de 00 21 07 df 00 27 07 e0 00 1c 
07 e1 00 1f 07 e2 01 b6 07 e3 00 1b 07 e4 00 1e 07 e5 02 0a 
07 e6 02 19 07 e7 02 19 07 e8 02 05 07 ee 3a 70 07 ef 3c 27 
07 f0 02 21 07 f1 02 4f 07 f2 01 d7 07 f3 39 d5 07 f4 37 9b 
07 f5 01 f0 07 f6 01 cc 07 f7 01 e8 07 f8 02 5c 07 f9 37 b3 
07 fa 01 fc e0 dd ab aa 
```

</details>

The data block is 928 bytes long.  Status info is in bytes 4 .. 923.  Each datum is 4 bytes long and consists of 2 2-byte integers.  First is a parameter code, second is the value.  We're looking for parameter code 1028 (0x0404) and either 113 (0x0071) or 144 (0x0090).  The controller ID is at the same offset in the data block of each packet.  Row 6, 2nd datum: `04 04 00 90`

> Write an asynInterpose layer

That sounds like a plan to implement a filter that should be most reliable.

prjemian, 2020-04-20:
Summary: `SO_REUSEPORT` socket option (the point of this issue) works.  Agree?

MarkRivers, 2020-04-20:
I don't think the asyn driver can access the packet header.  But it looks like what you need is the controller ID in the packet.  I think you should be able to write an interpose interface that filters messages based on that.  You could use asynInterposeFlush.c as a starting point.  But rather than overriding the flushIt() method you would override the readIt() method.

prjemian, 2020-04-20:
Thanks. That's the advice I needed next.

MarkRivers, 2020-05-05:
Closed via #109

closed 2020-05-05 by MarkRivers