epics-modules/asyn #110: broken link in documentation

prjemian, 2020-04-23:
On the main [documentation page](https://epics-modules.github.io/master/asyn/), the link to *How-to Guide for the Asyn Record* generates a 404 (not found) error.

The correct URL is: http://htmlpreview.github.io/?https://github.com/epics-modules/asyn/blob/master/documentation/HowToDoSerial/HowToDoSerial_StreamDevice.html

prjemian, 2020-04-23:
Where is the source of this document containing the broken link?

MarkRivers, 2020-04-23:
The missing document is here:

https://epics.anl.gov/modules/soft/asyn/AsynRecordControl/AsynRecordControl.html

When I moved the documentation from the APS to github.io I did not realize that document was not part of the asyn documentation, but rather was created outside of the asyn module.  I don't know if it is in some other repository.

The source of the file with the broken link is here:
https://github.com/epics-modules/epics-modules.github.io/blob/master/master/asyn/index.html

MarkRivers, 2020-05-06:
@anjohnson one of the links in the asyn documentation still points to a document on an APS web page:
https://epics.anl.gov/modules/soft/asyn/AsynRecordControl/AsynRecordControl.html
Is that file in a source code repository somewhere?  I don't think it was ever part of the asyn repository. 
I would like to get that file and add it to the asyn documentation, perhaps preserving its version history.

Are there other files in the directory pointed to by https://epics.anl.gov/modules/soft/asyn that should also be added to the asyn module?

anjohnson, 2020-05-06:
I looked in CVS, svn, and in a small cache of documents that Eric left me, with no sign of the sources for that particular document. There's only the one version of it in my website git repo, so I have no history to preserve. It isn't terribly long or complicated either; I just used pandoc to convert it to [an rst file](https://epics.anl.gov/modules/soft/asyn/AsynRecordControl/AsynRecordControl.rst) and to [GitHub-flavored Markdown](https://epics.anl.gov/modules/soft/asyn/AsynRecordControl/AsynRecordControl.md) – the rst looks like a better conversion but it's obviously up to you to decide how you want to import it.

The only other file in that directory is an old version of the StreamDevice HowToDoSerial doc which is already in the Asyn repo.

MarkRivers, 2020-07-17:
Fixed in R4-40.

closed 2020-07-17 by MarkRivers