epics-modules/asyn #112: Appveyor build not working

MarkRivers, 2020-05-05:
I am trying to add an appveyor builds to the .ci builds for asyn.

When I push to asyn it is triggering a build on appveyor, which is good.  However, it is failing with this error that I don't understand.  I did edit the .appveyor.yml file, but only to remove some old versions of Visual Studio, I don't think that would have caused this problem?
```
Error parsing appveyor.yml: (Line: 106, Col: 1, Idx: 2830) - (Line: 106, Col: 8, Idx: 2837): Duplicate key
```

ralphlange, 2020-05-05:
Will have a look tomorrow. No worries.

mark0n, 2020-05-05:
I'm suspecting that you are only allowed to have one "install" block. Have you tried removing line 19 + moving line 21 to the other "install" block in line 107?

MarkRivers, 2020-05-05:
Thanks @mark0n that worked to get past the first error.  This error is on master branch of ci-scripts in /appveyor/.appveyor.yml.example-full.

Now the appveyor build fails with this error:
```

Build started
git config --global core.autocrlf true
git clone -q --depth=50 --branch=master https://github.com/epics-modules/asyn.git C:\projects\asyn
git checkout -qf 8f236a6491aa2929322157ea46d49c191a0dcedd
Running Install scripts
git submodule update --init --recursive
Submodule '.ci' (https://github.com/epics-base/ci-scripts) registered for path '.ci'
Cloning into 'C:/projects/asyn/.ci'...
Submodule path '.ci': checked out 'a39346bc78b469f3ccc18ab7ad7cdce837d72a05'
python .ci/appveyor/do.py prepare
python: can't open file '.ci/appveyor/do.py': [Errno 2] No such file or directory
Command exited with code 2
Get-ChildItem *.tap -Recurse -Force | % { Push-AppveyorArtifact $_.FullName -FileName $_.Name }
python .ci/appveyor/do.py build test-results -s
python: can't open file '.ci/appveyor/do.py': [Errno 2] No such file or directory
```
I think this error is because it is checking out an older version of ci-scripts which predates support for appveyor.  

Remind me how I tell asyn what version of the ci-scripts submodule to use?

ralphlange, 2020-05-05:
https://github.com/epics-base/ci-scripts#frequently-asked-questions

MarkRivers, 2020-05-05:
@ralphlange thanks.  I realized my problem.  On my local asyn I had never done "git submodule update" so my .ci directory was empty.  That's actually OK I guess until one wants to update the version of the ci-scripts to be used.  It's working fine now.

MarkRivers, 2020-05-08:
I have changed .appveyor.yml per @ralphlange suggestions, and now the appveyor build succeeds.

It was a very useful thing to get running because it exposed the fact that the unittests are failing on all Windows dynamic builds, regardless of compiler.  I guess the unittests had never been run on dynamic Windows builds before.  I have reproduce the problem locally, and am opening a new issue for that.

closed 2020-05-08 by MarkRivers