epics-modules/asyn #113: unittests fail for Windows dynamic builds

MarkRivers, 2020-05-08:
Running the CI in appveyor exposed the fact that the unittests fail for dynamic builds on Windows for all compiler versions (VS 2019/2017/2015/2013, mingw).

I have reproduced the problem locally.  

The problem is that the int32cb callback function is not being called by asynPortClient when it should be.

This is the output on windows-x64-static:
```
ok 43 - client.read(&val)==asynParamUndefined
# int32cb() called with 55
ok 44 - client->read(&other)==asynSuccess
ok 45 - data==other
# int32cb() done
ok 46 - portA->callParamCallbacks()==asynSuccess
ok 47 - lastint32==55
ok 48 - cbcount==1
# int32cb() called with 43
ok 49 - client->read(&other)==asynSuccess
ok 50 - data==other
# int32cb() done
ok 51 - client.read(&val)==asynSuccess
ok 52 - val==43
ok 53 - lastint32==43
ok 54 - cbcount==2

    Results
    =======
       Tests: 54
      Passed:  54 = 100.00%
```

Note the diagnostic messages from int32cb, and all tests passed.

This is the output on windows-x64:
```
ok 43 - client.read(&val)==asynParamUndefined
ok 44 - portA->callParamCallbacks()==asynSuccess
not ok 45 - lastint32==55
not ok 46 - cbcount==1
ok 47 - client.read(&val)==asynSuccess
ok 48 - val==43
not ok 49 - lastint32==43
not ok 50 - cbcount==2

Planned 54 tests but only ran 50

    Results
    =======
       Tests: 50
      Passed:  46 = 92.00%
      Failed:   4 =  8.00%
```
Note that there are no diagnostic messages from int32cb and it is only executing 50 tests rather than 54.  This is because 2 of the tests are in the callback function, and that should be called twice.

I am not sure what is wrong.  I tried changing the code to add epicsShareFunc to the int32cb, but that did not help.
```
diff --git a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
index a26d79b..64b702d 100644
--- a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
+++ b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
@@ -16,6 +16,8 @@
 #include <asynPortDriver.h>
 #include <asynPortClient.h>

+#include <epicsExport.h>
+
 // fake out asynPortDriver callback dispatching
 extern "C" int interruptAccept;
 int interruptAccept=1;
@@ -32,7 +34,7 @@ typedef epicsGuard<asynPortDriver> Guard;
 epicsInt32 lastint32;
 size_t cbcount;

-void int32cb(void *userPvt, asynUser *pasynUser,
+epicsShareFunc void int32cb(void *userPvt, asynUser *pasynUser,
                                        epicsInt32 data)
 {
     testDiag("int32cb() called with %d", (int)data);
```

@mdavidsaver you wrote these tests.  Any idea why the int32cb function is not being called, but there are no error messages?

MarkRivers, 2020-05-08:
I have changed the unittests to turn on asynTrace on the port driver.

This is the output on windows-x64-static:
```
ok 42 - lastint32==1234
2020/05/08 08:38:05.646 portA asynManager::queueLockPort locking port
2020/05/08 08:38:05.646 asynPortDriver:getIntegerParam: port=portA error getting parameter 4 y, in list 0, value undefined
2020/05/08 08:38:05.646 portA queueUnlockPort
ok 43 - client.read(&val)==asynParamUndefined
# int32cb() called with 55
2020/05/08 08:38:05.646 portA asynManager::queueLockPort locking port
2020/05/08 08:38:05.646 asynPortDriver:readInt32: function=4, name=y, value=55
2020/05/08 08:38:05.646 asynInt32SyncIO read: 55
2020/05/08 08:38:05.647 portA queueUnlockPort
ok 44 - client->read(&other)==asynSuccess
ok 45 - data==other
# int32cb() done
ok 46 - portA->callParamCallbacks()==asynSuccess
ok 47 - lastint32==55
ok 48 - cbcount==1
2020/05/08 08:38:05.648 portA asynManager::queueLockPort locking port
# int32cb() called with 43
2020/05/08 08:38:05.648 portA asynManager::queueLockPort locking port
2020/05/08 08:38:05.648 asynPortDriver:readInt32: function=4, name=y, value=43
2020/05/08 08:38:05.648 asynInt32SyncIO read: 43
2020/05/08 08:38:05.648 portA queueUnlockPort
ok 49 - client->read(&other)==asynSuccess
ok 50 - data==other
# int32cb() done
2020/05/08 08:38:05.649 asynPortDriver:writeInt32: function=4, name=y, value=43
2020/05/08 08:38:05.649 asynInt32SyncIO wrote: 43
2020/05/08 08:38:05.649 portA queueUnlockPort
2020/05/08 08:38:05.649 portA asynManager::queueLockPort locking port
2020/05/08 08:38:05.649 asynPortDriver:readInt32: function=4, name=y, value=43
2020/05/08 08:38:05.649 asynInt32SyncIO read: 43
2020/05/08 08:38:05.649 portA queueUnlockPort
ok 51 - client.read(&val)==asynSuccess
ok 52 - val==43
ok 53 - lastint32==43
ok 54 - cbcount==2
2020/05/08 08:38:05.654 portA -1 cancelInterruptUser
2020/05/08 08:38:05.654 asynPortDriver:drvUserDestroy: this=0000015019366B90, pasynUser=0000015019369E68

    Results
    =======
       Tests: 54
      Passed:  54 = 100.00%
```

This is the output on windows-x64:
```
ok 41 - cbcount==0
ok 42 - lastint32==1234
2020/05/08 08:39:23.558 portA asynManager::queueLockPort locking port
2020/05/08 08:39:23.562 asynPortDriver:getIntegerParam: port=portA error getting parameter 4 y, in list 0, value undefined
2020/05/08 08:39:23.568 portA queueUnlockPort
ok 43 - client.read(&val)==asynParamUndefined
ok 44 - portA->callParamCallbacks()==asynSuccess
not ok 45 - lastint32==55
not ok 46 - cbcount==1
2020/05/08 08:39:23.570 portA asynManager::queueLockPort locking port
2020/05/08 08:39:23.573 asynPortDriver:writeInt32: function=4, name=y, value=43
2020/05/08 08:39:23.577 asynInt32SyncIO wrote: 43
2020/05/08 08:39:23.580 portA queueUnlockPort
2020/05/08 08:39:23.582 portA asynManager::queueLockPort locking port
2020/05/08 08:39:23.586 asynPortDriver:readInt32: function=4, name=y, value=43
2020/05/08 08:39:23.590 asynInt32SyncIO read: 43
2020/05/08 08:39:23.592 portA queueUnlockPort
ok 47 - client.read(&val)==asynSuccess
ok 48 - val==43
not ok 49 - lastint32==43
not ok 50 - cbcount==2
2020/05/08 08:39:23.595 portA -1 cancelInterruptUser
2020/05/08 08:39:23.598 asynPortDriver:drvUserDestroy: this=00000205761224A0, pasynUser=000002057612B908

Planned 54 tests but only ran 50

    Results
    =======
       Tests: 50
      Passed:  46 = 92.00%
      Failed:   4 =  8.00%
```
This does not shed much more light on the problem.  It just proves that callParamCallbacks() is not calling the registered callback function.

mdavidsaver, 2020-05-08:
My first suspicion is threading and (lack of) synchronization.  This code is assuming that `callParamCallbacks()` runs callbacks synchronously (`ASYN_CANBLOCK` not given).  I guess this is a wrong assumption?

https://github.com/epics-modules/asyn/blob/3f16b6e3f866b93564dac31e6a4428344013fafa/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp#L150-L154

MarkRivers, 2020-05-08:
I think the callbacks do run synchronously. I tried adding epicsThreadSleep(1.0) after the call to callParamCallbacks(), but that did not help. The fact that it always fails on dynamic builds and never on static builds, even with mingw, leads me to think it is not a race condition but rather something related to DLL exports.

I'll add some more debugging to see if I can track it down.

MarkRivers, 2020-05-08:
I figured out the problem.  The global variable interruptAccept is not being exported correctly on Windows dynamic builds.  I should be able to fix it.

MarkRivers, 2020-05-08:
I'm having trouble getting interruptAccept defined correctly.

This is my change which is based on dbAccess.c in base:
```
diff --git a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
index a26d79b..f7c4bb9 100644
--- a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
+++ b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
@@ -15,10 +15,10 @@

 #include <asynPortDriver.h>
 #include <asynPortClient.h>
+#include <epicsExport.h>

 // fake out asynPortDriver callback dispatching
-extern "C" int interruptAccept;
-int interruptAccept=1;
+epicsShareDef volatile int interruptAccept=1;

 #ifdef __rtems__
 // no test data needed (when running individual CI tests)
```
This builds and runs fine on Linux.

On windows-x64 it builds, but interruptAccept is 0 in callParamCallbacks.  This is the output with some debugging printf added.  Note that it says interruptAccept=0, while on Linux it says it is 1.
```
ok 38 - stdString == "Hello world"
ok 39 - portA->findParam(0, "y", &idx1)==asynSuccess
ok 40 - client.registerInterruptUser(&int32cb)==asynSuccess
paramList::callCallbacks addr=0 interruptAccept=0
ok 41 - cbcount==0
ok 42 - lastint32==1234
ok 43 - client.read(&val)==asynParamUndefined
paramList::callCallbacks addr=0 interruptAccept=0
ok 44 - portA->callParamCallbacks()==asynSuccess
not ok 45 - lastint32==55
not ok 46 - cbcount==1
paramList::callCallbacks addr=0 interruptAccept=0
ok 47 - client.read(&val)==asynSuccess
ok 48 - val==43
not ok 49 - lastint32==43
not ok 50 - cbcount==2

Planned 54 tests but only ran 50

    Results
    =======
       Tests: 50
      Passed:  46 = 92.00%
      Failed:   4 =  8.00%
```

The other problem is that the code now fails to compile on windows-x64-static:
```
link -nologo   -incremental:no -opt:ref -release  -MACHINE:X64             
 -out:asynPortDriverTest.exe   asynPortDriverTest.obj      
../../../../lib/windows-x64-static/asyn.lib  
H:/epics-devel/base-7.0.3.1/lib/windows-x64-static/Com.lib
netapi32.lib ws2_32.lib advapi32.lib user32.lib kernel32.lib 
winmm.lib dbghelp.lib asyn.lib(asynPortDriver.obj) 
: error LNK2019: unresolved external symbol interruptAccept referenced in function "public: virtual void __cdecl callbackThread::run(void)" (?run@callbackThread@@UEAAXXZ)
asynPortDriverTest.exe : fatal error LNK1120: 1 unresolved externals
make[1]: *** [H:/epics-devel/base-7.0.3.1/configure/RULES_BUILD:213: asynPortDriverTest.exe] Error 1120
make[1]: Leaving directory 'J:/epics/devel/asyn/asyn/asynPortDriver/unittest/O.windows-x64-static'
make: *** [H:/epics-devel/base-7.0.3.1/configure/RULES_ARCHS:58: install.windows-x64-static] Error 2
```
@anjohnson any ideas what I am doing wrong?

mdavidsaver, 2020-05-08:
```c++
#include <epicsExport.h>
...
extern "C" {
epicsShareExtern
int interruptAccept=1;
}
```
might work.  (`epicsShareDef` is a no-op these days)

MarkRivers, 2020-05-08:
I figured out the problem with the static builds was C++ name mangling, so I changed my patch to:
```
diff --git a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
index a26d79b..b0296b9 100644
--- a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
+++ b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
@@ -15,11 +15,12 @@

 #include <asynPortDriver.h>
 #include <asynPortClient.h>
+#include <epicsExport.h>

 // fake out asynPortDriver callback dispatching
-extern "C" int interruptAccept;
-int interruptAccept=1;
-
+extern "C" {
+  epicsShareDef volatile int interruptAccept=1;
+}
 #ifdef __rtems__
 // no test data needed (when running individual CI tests)
 const void* epicsRtemsFSImage = 0;
```
This now builds and works fine on linux-x86_64 and windows-x64-static.  But on windows-x64 (dynamic) interruptAccept is still 0 in asynPortDriver::callParamCallbacks().  

I am still doing something wrong.

anjohnson, 2020-05-08:
The original declaration of `interrupAccept` that you replaced was not marked as `volatile`, so that may be your mistake. I didn't think the `extern "C"` part was really needed for simple variable types, but that may have changed in recent years.

You might want to move the assignment into the top of your main() function instead of having it compiled-in.

MarkRivers, 2020-05-08:
I have a solution that works on windows-x64, windows-x64-static, and linux-x86_64.  It does the same thing as asynPortDriver.cpp and the top-level Makefile.  It is more future-proof because it builds the test application with everything from EPICS base unless EPICS_LIBCOM_ONLY is defined, as does the rest of asyn.
```
diff --git a/asyn/asynPortDriver/unittest/Makefile b/asyn/asynPortDriver/unittest/Makefile
index 7618c68..d4b76a6 100644
--- a/asyn/asynPortDriver/unittest/Makefile
+++ b/asyn/asynPortDriver/unittest/Makefile
@@ -11,7 +11,11 @@ TOP=../../..
 include $(TOP)/configure/CONFIG

 PROD_LIBS += asyn
-PROD_LIBS += Com
+ifeq ($(EPICS_LIBCOM_ONLY),YES)
+  PROD_LIBS += Com
+else
+  PROD_LIBS += $(EPICS_BASE_IOC_LIBS)
+endif

 #Tests for the parameters
 #TESTPROD_HOST += ParamValTest
diff --git a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
index a26d79b..1b22000 100644
--- a/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
+++ b/asyn/asynPortDriver/unittest/asynPortDriverTest.cpp
@@ -16,9 +16,12 @@
 #include <asynPortDriver.h>
 #include <asynPortClient.h>

-// fake out asynPortDriver callback dispatching
-extern "C" int interruptAccept;
-int interruptAccept=1;
+// Need interrupt accept from dbAccess.h unless asyn is built with EPICS_LIBCOM_ONLY
+#ifdef EPICS_LIBCOM_ONLY
+    static int interruptAccept;
+#else
+    #include <dbAccess.h>
+#endif

 #ifdef __rtems__
 // no test data needed (when running individual CI tests)
```

closed 2020-05-08 by MarkRivers