epics-modules/asyn #118: Compilation problem on master 31df0435853d3125bab5

tboegi, 2020-06-18:
Trying to compile asyn on this commit
commit 31df0435853d3125bab5b9bb4ac51c3ca81cdd89 (HEAD, origin/master, origin/HEAD)
Merge: 13716709 158ca6bc
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Tue Jun 16 10:49:36 2020 -0500
    Merge pull request #114 from krisztianloki/master
    Take connect timestamp after connection attempt
Gives the following error message, please see below.

(Probably the commit did not introduce the problem.
 I hope, that it is easy to fix by an expert 

[snip]
/usr/bin/ar -rc    libtestConnectSupport.a     testConnect.o
/usr/bin/ranlib libtestConnectSupport.a
c++ -o libtestConnectSupport.dylib -dynamiclib -flat_namespace -undefined dynamic_lookup -install_name /home/epics/modules/asyn/lib/darwin-x86/libtestConnectSupport.dylib   -L/home/epics/modules/asyn/lib/darwin-x86 -L/home/epics/base/lib/darwin-x86        -arch x86_64             testConnect.o   -lasyn -ldbRecStd -ldbCore -lca -lCom      -lreadline -lm 
Installing shared library ../../../lib/darwin-x86/libtestConnectSupport.dylib
Installing library ../../../lib/darwin-x86/libtestConnectSupport.a
c++              -DUNIX  -Ddarwin     -O3 -g   -Wall       -arch x86_64    -fno-common  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/clang -I../../../include/os/Darwin -I../../../include -I/home/epics/base/include/compiler/clang -I/home/epics/base/include/os/Darwin -I/home/epics/base/include   -I/home/epics/base/../modules/asyn/include                  -I/home/epics/base/../modules/calc/include                                                                                                  -c testConnect_registerRecordDeviceDriver.cpp
c++              -DUNIX  -Ddarwin     -O3 -g   -Wall       -arch x86_64    -fno-common  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/clang -I../../../include/os/Darwin -I../../../include -I/home/epics/base/include/compiler/clang -I/home/epics/base/include/os/Darwin -I/home/epics/base/include   -I/home/epics/base/../modules/asyn/include                  -I/home/epics/base/../modules/calc/include                                                                                                  -c ../testConnectMain.cpp
c++ -o testConnect  -L/home/epics/modules/asyn/lib/darwin-x86 -L/home/epics/base/lib/darwin-x86        -arch x86_64             testConnect_registerRecordDeviceDriver.o testConnectMain.o   -ltestConnectSupport -lasyn -ldbRecStd -ldbCore -lca -lCom   
Installing created executable ../../../bin/darwin-x86/testConnect
/Library/Developer/CommandLineTools/usr/bin/make -C ./testEpicsApp install 
/Library/Developer/CommandLineTools/usr/bin/make -C ./src install 
perl -CSD /home/epics/base/bin/darwin-x86/makeMakefile.pl O.darwin-x86 ../../..
mkdir -p O.Common
/Library/Developer/CommandLineTools/usr/bin/make -C O.darwin-x86 -f ../Makefile TOP=../../.. \
	    T_A=darwin-x86 install
perl -CSD /home/epics/base/bin/darwin-x86/mkmf.pl  -m uint32DigitalDriver.d -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/clang -I../../../include/os/Darwin -I../../../include -I/home/epics/base/include/compiler/clang -I/home/epics/base/include/os/Darwin -I/home/epics/base/include   -I../../../include                  -I/home/epics/base/../modules/calc/include                                                                                                  uint32DigitalDriver.o ../uint32DigitalDriver.c
perl -CSD /home/epics/base/bin/darwin-x86/mkmf.pl  -m int32Driver.d -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/clang -I../../../include/os/Darwin -I../../../include -I/home/epics/base/include/compiler/clang -I/home/epics/base/include/os/Darwin -I/home/epics/base/include   -I../../../include                  -I/home/epics/base/../modules/calc/include                                                                                                  int32Driver.o ../int32Driver.c
perl -CSD /home/epics/base/bin/darwin-x86/mkmf.pl  -m testEpicsMain.d -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/clang -I../../../include/os/Darwin -I../../../include -I/home/epics/base/include/compiler/clang -I/home/epics/base/include/os/Darwin -I/home/epics/base/include   -I../../../include                  -I/home/epics/base/../modules/calc/include                                                                                                  testEpicsMain.o ../testEpicsMain.cpp
Creating dbd file testEpics.dbd
perl -CSD /home/epics/base/bin/darwin-x86/dbdExpand.pl   -I. -I.. -I../O.Common -I../../../dbd -I/home/epics/base/dbd -I../../../dbd -I/home/epics/base/../modules/calc/dbd -o testEpics.dbd testEpicsCommon.dbd calcSupport.dbd asynCalc.dbd
dbdExpand.pl: Can't find file 'asynCalc.dbd'
  while reading 'asynCalc.dbd' to create 'testEpics.dbd'
  Your Makefile may need this dependency rule:
    $(COMMON_DIR)/testEpics.dbd: $(COMMON_DIR)/asynCalc.dbd
dbdExpand.pl: Exiting due to errors
make[3]: *** [../O.Common/testEpics.dbd] Error 2
make[2]: *** [install.darwin-x86] Error 2
make[1]: *** [src.install] Error 2
make: *** [testEpicsApp.install] Error 2

MarkRivers, 2020-06-18:
I believe that your problem is that you have uncommented this line in asyn/configure/RELEASE
```
#CALC=$(SUPPORT)/calc-3-7-3
```

But you have not uncommented this line:
```
#SSCAN=$(SUPPORT)/sscan-2-11-3
```

If I do that I can reproduce your error. 

That is happening because asyn/Makefile contains this:
```
  # do we have EPICS calc module (scalcout etc.)
  # applications should include asynCalc.dbd as well as asyn.dbd if support is needed
  ifdef CALC
  ifdef SSCAN
    USR_CFLAGS += -DHAVE_CALCMOD
    ASYN_LIBS += calc sscan
    DBD += asynCalc.dbd
  endif
  endif
```
So it only installs asynCalc.dbd if *both* CALC and SSCAN are defined.  That is because there was an assumption that CALC is always built with SSCAN in order to support the swait record.  This is not necessarily true, so I will change the Makefile to separate the dependencies on CALC and SSCAN and see if that works.

tboegi, 2020-06-18:
Thanks for digging.
After having problems to get rid off the SUPPORT macro in configure/RELEASE,
I have overwritten configure/RELEASE with a version containing only one line:

```
>  cat RELEASE
> include $(TOP)/configure/RELEASE_PATHS.local.$(EPICS_HOST_ARCH)
```
And configure/RELEASE_PATHS.local.darwin-x86 looks like this: 

```
SSCAN=
CALC        = $(EPICS_BASE)/../modules/calc
```

MarkRivers, 2020-06-18:
I have pushed these commits:

```

corvette:~/devel/asyn/asyn>git log
commit 8573487454c20f6698f144ac68eb51ae68a22450
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Thu Jun 18 08:57:08 2020 -0500

    Remove SSCAN dependency, it is not needed for building library, only for testEpicsApp application

commit 34e11c7b86ed8ca043163e11cd330a52663b13ce
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Thu Jun 18 08:56:14 2020 -0500

    Added comment about when SSCAN is needed

commit 677fa843a02f4afe247889c9ad838c336c692b7f
Author: Mark Rivers <rivers@cars.uchicago.edu>
Date:   Thu Jun 18 08:55:18 2020 -0500

    Separate CALC and SSCAN dependencies.  SSCAN is needed only if CALC is built with SSCAN support
```
It now works to build with CALC defined but SSCAN not defined.  

On Linux if CALC is built with SSCAN support testEpicsApp runs OK even if SSCAN is not defined, because the libsscan.so is found at run-time.  However, on dynamic Windows build it will link OK but generate a run-time error because sscan.dll cannot be found.  So it is best to define SSCAN if CALC is built with SSCAN.

tboegi, 2020-06-18:
Thank you very much @MarkRivers for the fast response.
This one fixes my problems:
https://github.com/epics-modules/asyn/commit/8573487454c20f6698f144ac68eb51ae68a22450

closed 2020-06-18 by tboegi