epics-modules/asyn #132: Unable to compile on Fedora 33 due to

thomasms, 2021-03-16:
I am getting the following error when building from source (version R4.41).
```
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE               -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_DEVINT64  -mtune=generic     -m64 -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include   -I/home/user/Dev/ipac/include   -I/home/user/Dev/epics-sequencer/seq-2.2.6/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include -I/home/user/Dev/epics-base/include/compiler/gcc -I/home/user/Dev/epics-base/include/os/Linux -I/home/user/Dev/epics-base/include        -c ../../asyn/asynRecord/drvAsyn.c

make[2]: *** No rule to make target 'vxi11core_xdr.c', needed by 'vxi11core_xdr.o'.  Stop.
make[2]: Leaving directory '/home/user/Dev/asyn/asyn/O.linux-x86_64'
make[1]: *** [/home/user/Dev/epics-base/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/user/Dev/asyn/asyn'
make: *** [/home/user/Dev/epics-base/configure/RULES_DIRS:85: asyn.install] Error 2
```

```bash
# in the async/Makefile
SRC_DIRS += $(ASYN)/vxi11
ifeq ($(OS_CLASS), WIN32)
  asyn_SRCS += drvVxi11Win32.c
else
  asyn_SRCS += vxi11core_xdr.c
  asyn_SRCS += drvVxi11.c
endif
```
Changing this to the correct path for `vxi11core_xdr.c` I get a different error to do with rpc/rpc.h:

```bash
# in the async/Makefile
SRC_DIRS += $(ASYN)/vxi11
ifeq ($(OS_CLASS), WIN32)
  asyn_SRCS += drvVxi11Win32.c
else
  # this line changed
  asyn_SRCS += rpc/vxi11core_xdr.c
  asyn_SRCS += drvVxi11.c
endif
```

The error is:
```bash
In file included from ../../asyn/vxi11/rpc/vxi11core_xdr.c:6:
../../asyn/vxi11/rpc/vxi11core.h:9:10: fatal error: rpc/rpc.h: No such file or directory
    9 | #include <rpc/rpc.h>
      |          ^~~~~~~~~~~
compilation terminated.

make[2]: *** [/home/user/Dev/epics-base/configure/RULES_BUILD:255: rpc/vxi11core_xdr.o] Error 1
make[2]: Leaving directory '/home/user/Dev/asyn/asyn/O.linux-x86_64'
make[1]: *** [/home/user/Dev/epics-base/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/user/Dev/asyn/asyn'
make: *** [/home/user/Dev/epics-base/configure/RULES_DIRS:85: asyn.install] Error 2

```

Any ideas on what is going on here? I am building from source following the steps stated here: https://epics-modules.github.io/master/asyn/ 

On Fedora 33 using:
- gcc 10.2.1
- perl 5.32.1
- epics-base 7.0.4
- sequencer 2.2.6
- ipac 2.16

MarkRivers, 2021-03-16:
The issue is that Fedora 33 uses the tirpc package for rpc.  With that package rpc.h is in /usr/include/tirpc/rpc/rpc.h, rather than in /usr/include/rpc/rpc.h.

That means you need to uncomment the line TIRPC=YES in asyn/configure/CONFIG_SITE.
```
# Some linux systems moved RPC related symbols to libtirpc
# To enable linking against this library, uncomment the following line
# TIRPC=YES
```
Your change to the Makefile is also due to the same reason, because it is not finding rpcgen on your system.  Once it does then vxi11core_xdr.c will be created in the O.linux-x86_64 directory at build time using rpcgen.  The version in the rpc/ directory should not be used except for certain cross-compiles, look at the README in that directory.

The use of tirpc and the need to edit CONFIG_SITE is documented in the [release notes for R4-34](https://epics-modules.github.io/master/asyn/R4-34/RELEASE_NOTES.html).

thomasms, 2021-03-16:
Thanks for the tip but unfortunately that doesn't solve the issue. I uncomment the line as you suggest,
```diff
diff --git a/configure/CONFIG_SITE b/configure/CONFIG_SITE
index 538cd55c..1cd8a4b1 100644
--- a/configure/CONFIG_SITE
+++ b/configure/CONFIG_SITE
@@ -56,7 +56,7 @@ DRV_FTDI_USE_LIBFTDI1=NO
 
 # Some linux systems moved RPC related symbols to libtirpc
 # To enable linking against this library, uncomment the following line
-# TIRPC=YES
+TIRPC=YES

 -include $(SUPPORT)/configure/CONFIG_SITE
```
I now get the following error (following a `make clean uninstall`):
```make
cp ../vxi11/vxi11intr.rpcl .
rpcgen  vxi11intr.rpcl
make[2]: rpcgen: No such file or directory
cp ../vxi11/vxi11core.rpcl .
rpcgen  vxi11core.rpcl
make[2]: rpcgen: No such file or directory
make[2]: *** No rule to make target 'vxi11core_xdr.c', needed by 'vxi11core_xdr.o'.  Stop.
make[2]: Leaving directory '/home/user/Dev/asyn/asyn/O.linux-x86_64'
make[1]: *** [/home/user/Dev/epics-base/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/user/Dev/asyn/asyn'
make: *** [/home/user/Dev/epics-base/configure/RULES_DIRS:85: asyn.install] Error 2
```
It turns out I had libtirpc but not libtirpc-devel (so no headers in either directory), but upon installing the latter, I do indeed have headers at:
```bash
➜  asyn git:(0c70f16c) ✗ ll /usr/include/tirpc 
total 4.0K
-rw-r--r--. 1 root root 2.2K Aug  4  2020 netconfig.h
drwxr-xr-x. 1 root root  556 Mar 16 14:03 rpc
drwxr-xr-x. 1 root root   28 Mar 16 14:03 rpcsvc
```

Anything else I am missing?

tboegi, 2021-03-16:
rpcgen is an executable binary - 
you need to install it as part of some rpm - don't know how it is called

type rpcgen 
rpcgen is /usr/bin/rpcgen

thomasms, 2021-03-16:
OK, great. That helps but I get further down the rabbit hole....

```make
make[3]: *** No rule to make target 'ipSNCServer.c', needed by 'ipSNCServer.o'.  Stop.
```

What am I missing now?

tboegi, 2021-03-16:
This is part of the test...buisiness. I only have these files- not sure if this helps:

./asyn $ find . -name ipSNCServer*
./testIPServerApp/adl/ipSNCServer.adl
./testIPServerApp/src/ipSNCServer.st

MarkRivers, 2021-03-16:
> What am I missing now?

That error is coming because it is trying to compile an SNL program but it cannot find the SNL compiler.  That is happening because your asyn/configure/RELEASE has this line uncommented:
```
SNCSEQ=$(SUPPORT)/seq-2-2-5
```
You have 2 choices:
- Comment out that line.
- Build the seq module and define its location in that line.

thomasms, 2021-03-16:
Ah, that was it! I had cloned the code for seq-2.2.6 but for some reason it never compiled properly the first time.
After recompiling seq-2.6.6 I was able to now build asyn.

Thank you for help, but I do feel like this should be better documented. Even if it is something like: "Note this requires libtirrpc-devel & rpc installed.....".
Also, the point about uncommenting TIRPC would be useful to know, other than buried in the release notes.

MarkRivers, 2021-03-16:
> Thank you for help, but I do feel like this should be better documented. Even if it is something like: "Note this requires libtirrpc-devel & rpc installed.....".

Actually I have documented all of the packages required not only for asyn, but for many of the EPICS modules:
https://docs.epics-controls.org/projects/how-tos/en/latest/getting-started/linux-packages.html

MarkRivers, 2021-03-16:
> Also, the point about uncommenting TIRPC would be useful to know, other than buried in the release notes.

It is documented in CONFIG_SITE.  As its name implies it is a site-specific file that you should always look at when building a new module to see if there are site-specific customizations required.  

Note that TIRPC=YES is only required for certain Linux distributions, and I don't know of a good way to document when it is or is not required.  It is not required on RHEL 7, for example.

thomasms, 2021-03-16:
OK, fair enough. Indeed the documentation is there, just unbeknownst to me that it resided in a different location.
Maybe providing a link to the link you mentioned on the README, or on the Github pages, would be helpful.

With regard to the CONFIG_SITE, you're correct - it is stated there, it is just not clear to me (fairly new to EPICS) where I should be looking, but maybe this is just me.

All in all, it's just about saving time as a user/developer, but I am very grateful for your prompt help in pointing these matters out to me.

closed 2021-03-16 by thomasms