epics-modules/asyn #138: typo or wrong calling for outputCallbackCallback in devEpics modules?

hyteng, 2021-09-05:
many asynInterface devSet modules in devEpics , such as devAsynInt32.c, there is one function outputCallbackCallback for activating the second process of a PVRecord:

static void outputCallbackCallback(CALLBACK *pcb)
{
    static const char *functionName="outputCallbackCallback";

    devPvt *pPvt;
    callbackGetUser(pPvt, pcb);
    {
        dbCommon *pr = pPvt->pr;
        dbScanLock(pr);
        epicsMutexLock(pPvt->devPvtLock);
        pPvt->newOutputCallbackValue = 1;
        dbProcess(pr);
        if (pPvt->newOutputCallbackValue != 0) {
            /* We called dbProcess but the record did not process, perhaps because PACT was 1
             * Need to remove ring buffer element */
            asynPrint(pPvt->pasynUser, ASYN_TRACE_ERROR,
                "%s %s::%s warning dbProcess did not process record, PACT=%d\n",
                pr->name, driverName, functionName,pr->pact);
            getCallbackValue(pPvt);
            pPvt->newOutputCallbackValue = 0;
        }
        epicsMutexUnlock(pPvt->devPvtLock);
        dbScanUnlock(pr);
    }
}

this function is also registered in commonInit function:

static long initCommon(dbCommon *pr, DBLINK *plink,
    userCallback processCallback,interruptCallbackInt32 interruptCallback, interruptCallbackEnum callbackEnum,
    int maxEnums, char *pFirstString, int *pFirstValue, epicsEnum16 *pFirstSeverity)
{
    ...   
/* If the info field "asyn:READBACK" is 1 and interruptCallback is not NULL
     * then register for callbacks on output records */
    if (interruptCallback) {
        int enableCallbacks=0;
        ...
        if (enableCallbacks) {
            status = createRingBuffer(pr);
            if (status!=asynSuccess) goto bad;
            status = pPvt->pint32->registerInterruptUser(
               pPvt->int32Pvt,pPvt->pasynUser,
               pPvt->interruptCallback,pPvt,&pPvt->registrarPvt);
            if (status!=asynSuccess) {
                printf("%s %s::%s error calling registerInterruptUser %s\n",
                       pr->name, driverName, functionName,pPvt->pasynUser->errorMessage);
            }
            /* Initialize the interrupt callback */
            callbackSetCallback(outputCallbackCallback, &pPvt->outputCallback);
            callbackSetPriority(pr->prio, &pPvt->outputCallback);
            callbackSetUser(pPvt, &pPvt->outputCallback);
        }
    }
    return INIT_OK;
bad:
    recGblSetSevr(pr,LINK_ALARM,INVALID_ALARM);
    pr->pact=1;
    return INIT_ERROR;
}

so the function is registered and assigned into pPvt->outputCallback. but in the port thread responding to
 queueRequest, such as punction rocessCallbackInput and processCallbackOutput, the callbackRequestProcessCallback usees another parameter pPvt->processCallback, which seems never get registered or assignment else where.

static void processCallbackInput(asynUser *pasynUser)
{
    ...
    pPvt->lastStatus = pPvt->result.status;
    if(pr->pact) callbackRequestProcessCallback(&pPvt->processCallback,pr->prio,pr);
}

static void processCallbackOutput(asynUser *pasynUser)
{
    ...
    pPvt->lastStatus = pPvt->result.status;
    if(pr->pact) callbackRequestProcessCallback(&pPvt->processCallback,pr->prio,pr);
}   

should their callbackRequestProcessCallback function take &pPvt->outputCallback as the 1st parameter? or is there other tricks?

MarkRivers, 2021-09-05:
> but in the port thread responding to queueRequest, such as function processCallbackInput and processCallbackOutput, the callbackRequestProcessCallback uses another parameter pPvt->processCallback, which seems never get registered or assignment else where.

pPvt->processCallback is initialized here 
https://github.com/epics-modules/asyn/blob/64b8a71b9dfbe36e75da03427f151e9b2c7016a9/asyn/devEpics/devAsynInt32.c#L519
and here
https://github.com/epics-modules/asyn/blob/64b8a71b9dfbe36e75da03427f151e9b2c7016a9/asyn/devEpics/devAsynInt32.c#L544

This is explained in the EPICS Developer's Guide (https://epics.anl.gov/base/R3-16/1-docs/AppDevGuide/EPICSGeneralPurposeTasks.html) which says:
```
callbackRequestProcessCallback issues the calls:
    callbackSetCallback(ProcessCallback, pCallback); 
    callbackSetPriority(Priority, pCallback); 
    callbackSetUser(pRec, pCallback); 
    callbackRequest(pCallback);
```
So the pointer pCallback which is passed as the first argument to callbackRequestProcessCallback() is actually initialized by that function, and does not need to have been previously initialized.  The callback is always to the pr->process() function.

MarkRivers, 2021-09-05:
Are you observing any behavior which makes you think the asyn device support is not working correctly?

hyteng, 2021-09-06:
Oh, that is it, I was developing my asyn modules and got puzzled about how the devset modules call for a second process. Actually there is another ProcessCallback function for it. So the code works correctly, thanks for tips.

hyteng, 2021-09-06:
> 
> 
> Are you observing any behavior which makes you think the asyn device support is not working correctly?

No, actually it's just my puzzle, sorry for the trouble.