epics-modules/asyn #139: asynXXXArray should support aai and aao record types

ralphlange, 2021-09-07:
asyn only supports the waveform record for array data.

The aai and aao records have been part of EPICS Base for a long time. They provide proper input and output functionality for array data and are similar enough to the waveform record to be supported without much effort.

vnadot, 2022-03-23:
Hi @ralphlange ,

I'm indeed looking for a way to write an array with asyn (for my FPGA). What is your trick without this asyn feature ?

The array read works fine for me:
```
record(waveform, "$(P=CEA:)FileReader0DataRb") {
    field(DESC, "File reader 0 data")
    field(DTYP, "asynFloat64ArrayIn")
    field(FTVL, "DOUBLE")
    field(NELM, "${NELM=4096}")
    field(SCAN, "I/O Intr")
    field(INP,  "@asyn(${PORT})FileReader0Data")
}
```

ralphlange, 2022-03-23:
Asyn also supports `asyn<TYPE>ArrayOut` for the waveform record.
As simple as that.

vnadot, 2022-03-23:
Indeed, it works fine like this, thanks !

```
record(waveform, "${P=CEA:}FileReader0DataSet") {
    field(DESC, "File reader 0 data")
    field(DTYP, "asynFloat64ArrayOut")
    field(FTVL, "DOUBLE")
    field(NELM, "${NELM=4096}")
    field(INP,  "@asyn(${PORT})FileReader0Data")
    field(EGU,  "${EGU=V}") # should be set by saverestoreApp ?
    field(PREC, "${PREC=3}")
}
record(waveform, "${P=CEA:}FileReader0DataRb") {
    field(DESC, "File reader 0 data")
    field(DTYP, "asynFloat64ArrayIn")
    field(FTVL, "DOUBLE")
    field(NELM, "${NELM=4096}")
    field(SCAN, "I/O Intr")
    field(INP,  "@asyn(${PORT})FileReader0Data")
    field(EGU,  "${EGU=V}") # should be set by saverestoreApp ?
    field(PREC, "${PREC=3}")
}
```

ralphlange, 2022-03-23:
Sure.
It's more an annoyance that EPICS Base has proper array-in and array-out records, but asyn ignores them. Under the hood, it's all there.

MarkRivers, 2022-03-27:
I would like to add support for aao and aai.  But I want to do it without needing to write a large amount new code that is repetitive with the waveform record support.

Please look at the existing asyn device support code for the waveform record.  It consists of a 485 line C macro ASYN_XXX_ARRAY_FUNCS in this file:
https://github.com/epics-modules/asyn/blob/master/asyn/devEpics/devAsynXXXArray.h

That macro is invoked 6 times by the device support C files for Int8, Int16, Int32, Int64, Float32, and Float64 data types, e.g.
https://github.com/epics-modules/asyn/blob/master/asyn/devEpics/devAsynInt8Array.c

That mechnism is not very elegant, but it saves a huge amount of repetitive code.  Note that for the integer data types it supports both the signed and unsigned data types, so it supports 12 different values of the waveform record FTVL field.  The question is whether there is a better way to do it, and how to add support for aao and aai without increasing the code size by 2X-3X.

Could C++ template functions be used instead?  Note that the macro defines multiple functions, and also defines the DSETS.  It is not clear to me if that can be done in template functions.

@mdavidsaver I'd appreciate any advice you might have on this.

mdavidsaver, 2022-03-30:
> Could C++ template functions be used instead?

Certainly.  I've been doing this for some years now.  This can be done without requiring any modern c++ features.

mdavidsaver, 2022-03-30:
One real world example: https://github.com/mdavidsaver/pscdrv/blob/36c48e233915597c9cc860b9d587836a17e5a8ce/coreApp/src/devPSCSingle.cpp#L269-L285

ralphlange, 2022-04-08:
Loosely related, but maybe interesting as you are handling arrays.

I did a C++ Device Support for an FFT library last year, and I used it to create a prototype for using C++ smart pointers that are exposing their array data to the EPICS record directly. Minimal amount of copying, and multiple array input records can show different parts of a large array (sort of ROI) by pointing to different parts of the same shared array instance (with almost zero overhead).

This uses a "connector" class to hold the shared pointer and manage the BPTR fields of the related records.
https://github.com/ralphlange/fftw/blob/5e63ebd54377bff9d754a2a8fbe40b38ef030516/fftwDevSup/fftwConnector.cpp#L99

MarkRivers, 2022-09-30:
Closed via #162

closed 2022-09-30 by MarkRivers