epics-modules/asyn #143: hanging format specifiers

krmpotic, 2021-10-04:
While looking at `asyn/devEpics/devAsynFloat64.c` I noticed callocMustSucceed is passed a msg argument with a format specifier `%s`, but I don't see where it would get expanded.

see:
https://github.com/epics-modules/asyn/blob/master/asyn/devEpics/devAsynFloat64.c#L147
https://github.com/epics-modules/asyn/blob/master/asyn/devEpics/devAsynFloat64.c#L270
https://github.com/epics-modules/asyn/blob/master/asyn/devEpics/devAsynUInt32Digital.c#L338

MarkRivers, 2021-10-04:
Those lines should all be changed to be like this:
```
        pPvt->ringBuffer = callocMustSucceed(pPvt->ringSize+1, sizeof *pPvt->ringBuffer, "%s::%s", driverName, functionName);
````
@krmpotic can you do that and make a PR?

anjohnson, 2021-10-04:
@MarkRivers Assuming the `callocMustSucceed()` that these are calling is the one from Base's cantProceed.h header, that won't work. The `cantProceed()` routine does allow printf-style parameters, but both `callocMustSucceed()` and `mallocMustSucceed()` only accept a single `const char *` for the error message, no parameters.

MarkRivers, 2021-10-04:
@anjohnson thanks for pointing that out. In that case they should just be changed to strings like the following:
```
pPvt->ringBuffer = callocMustSucceed(pPvt->ringSize+1, sizeof *pPvt->ringBuffer, "devAsynFloat64::createRingBuffer");
```
where "devAsynFloat64" and "createRingBuffer" are changed as appropriate for the actual file and function.

krmpotic, 2021-10-04:
@MarkRivers sure, I can do a PR. I'll do it as you suggest.

krmpotic, 2021-10-04:
see https://github.com/epics-modules/asyn/pull/144

P.S. while checking if there are any other cases of this error (I checked all *MustSucceed calls, others are fine), I noticed that *MustSucceed sometimes use the `xxx::yyy` format and sometimes the `xxx:yyy` format. Is there are deeper reason for that, or can we unify it?

MarkRivers, 2021-10-04:
I think we should use the xxx::yyy syntax to be similar C++ syntax.

krmpotic, 2021-10-04:
Sounds good, I'll do that, and try not leave any one out.

closed 2021-10-05 by krmpotic