epics-modules/asyn #164: Unable to use drvAsynSerial with setOption baud rate 28800

edecats, 2022-10-20:
Hi - release notes indicate support was added for baud 28800 back in 4.9 but trying to set this option produces this failure -

`## Open Asyn port to RGA
drvAsynSerialPortConfigure ("asyn_rga", "/dev/ttyUSB0")
asynSetOption("asyn_rga",0, "baud", "28800")
setOption failed Unsupported data rate (28800 baud)`

You can see here a timeout from the device (most certainly due to incorrect baud rate setting as using pyserial with the same command results in a sucess) - unfortunately the SRS residual gas analyzer ONLY supports 28800 baud communication to my understanding. Was 28800 never actually implemented or am I using asyn set option incorrectly somehow? thanks so much!

`iocRun: All initialization complete
epics> 2022/10/20 16:53:31.203967 asyn_rga rga_ModelNumber_RB: No reply within 1000 ms to "ID?<0d>"`

MarkRivers, 2022-10-21:
If you look at the code here you will see that 28800 is supported if it is defined by the host.
https://github.com/epics-modules/asyn/blob/299be139652ff6a4528bb27cd8a65bd160fb0ad6/asyn/drvAsynSerial/drvAsynSerialPort.c#L285

I think that definition would come from termios.h.

edecats, 2022-10-21:
Hi Mark, 
Thanks for the quick response. Do you know if 28800 baud has ever been tested with asyn? I know that from a hardware standpoint the architecture supports this baudrate but I'm not really sure how I would even go about using it with a Linux OS -for instance when I look in usr/include/termios.h I don't see anywhere that I would be able to just define another arbitrary baudrate - it seems as though any fix would involve recompiling at a kernel level - am I missing something?

MarkRivers, 2022-10-22:
I also did not find anything in any include file in /usr/include or below listing baud rates like in the asyn driver. Perhaps Linux has changed how this is handled?

MarkRivers, 2022-10-22:
I retract my last comment. When I look in /usr/include/asm-generic/termbits.h I see B9600, B19200, and many higher baud rates, but not B28800. This is on Centos 7. This is why asyn is giving the error you see. Maybe there is a way to specify that baud rate even if it is not in termbits.h?

MarkRivers, 2022-10-22:
I've done a little Googling and 28800 is a "non-standard" baud rate that is not supported in termios.h.  It appears that there may some ways to make it work.

https://stackoverflow.com/questions/12646324/how-can-i-set-a-custom-baud-rate-on-linux
https://github.com/dcuddeback/serial-rs/issues/49

MarkRivers, 2022-10-22:
I just looked at the baud rates available on a Moxa terminal server, and it also does not support 28800.
https://shopmoxa.neteon.net/content/pdf/moxa-nport-5400-series-datasheet-v1.8.pdf

edecats, 2022-10-24:
Hi Mark -

Interesting, thanks for this - so I put a quick hacky modification into drvAsynSerial with that approach you linked - something like this:

`#include <asm/termios.h>
#include <stdio.h>
#include <fcntl.h>
#include <sys/stat.h>
#include <errno.h>
#include <asm/ioctls.h>
#include "drvAsynSerialCustomBaud.h"

void baud_custom(int rate, int fd){
    struct termios2 tio;
    ioctl(fd, TCGETS2, &tio);
    tio.c_cflag &= ~CBAUD;
    tio.c_cflag |= BOTHER;
    tio.c_ispeed = rate;
    tio.c_ospeed = rate;
    printf("Baud rate setting returned %i", ioctl(fd, TCSETS2, &tio));
}
`
where baud_custom gets run in the 28800 baud case in drvAsynSerialPort. It runs without errors and seems like it should produce the desired result - but it still times out when reading from the device. Stty on the USB port reveals a baud rate of 0, and then a baud rate of 9600 after ioc termination... I suspect this might be responsible: https://github.com/torvalds/linux/blob/337a0a0b63f1c30195733eaacf39e4310a592a68/drivers/usb/serial/pl2303.c#L598-L599

Unfortunately I think all of our Serial-USB converters probably use this same prolific chipset. Since the 28800 baudrate is not an issue on Windows (so far as I can tell), I'll likely have to rebuild my IOC on windows >.< anyway, thanks again for all the support!

edecats, 2022-10-26:
Migration onto windows has resolved the issue - closing. thanks!

closed 2022-10-26 by edecats