epics-modules/asyn #165: Is this asyn tutorial still correct?

LeeYangLBLBCS, 2022-10-26:
I tried to follow this document:
https://epics.anl.gov/modules/soft/asyn/R4-23/HowToDoSerial/HowToDoSerial_StreamDevice.html
which can be found in 
synApps/asyn-R4-43/documentation/HowToDoSerial/HowToDoSerial_StreamDevice.html
Epics base 7.0.6.1
asyn-R4-43
StreamDevice-2-2-23

The generated files failed to compile.

Is this tutorial too old?

here is the error I got:

make -C ./src install
make[2]: Entering directory '/home/lyang/StreamDeviceHowTo/keithley6517TestApp/src'
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. \
    T_A=linux-x86_64 install
make[3]: Entering directory '/home/lyang/StreamDeviceHowTo/keithley6517TestApp/src/O.linux-x86_64'
Creating dbd file keithley6517Test.dbd
perl -CSD /opt/epics_7.0.6.1/base-7.0.6.1/bin/linux-x86_64/dbdExpand.pl   -I. -I.. -I../O.Common -I../../../dbd -I/opt/epics/modules/support/asyn-R4-43/dbd -I/opt/epics/modules/support/StreamDevice-2-8-23/dbd -I/opt/epics_7.0.6.1/base-7.0.6.1/dbd -o keithley6517Test.dbd base.dbd stream.dbd asyn.dbd drvAsynSerialPort.dbd
dbdExpand.pl: Device 'stream' refers to unknown record type 'scalcout'.
	DBD files must be combined in the correct order.
Context: file '/opt/epics/modules/support/StreamDevice-2-8-23/dbd/stream.dbd'
  while reading 'stream.dbd' to create 'keithley6517Test.dbd'
dbdExpand.pl: Exiting due to errors
make[3]: *** [/opt/epics_7.0.6.1/base-7.0.6.1/configure/RULES.Db:405: ../O.Common/keithley6517Test.dbd] Error 255
make[3]: Leaving directory '/home/lyang/StreamDeviceHowTo/keithley6517TestApp/src/O.linux-x86_64'
make[2]: *** [/opt/epics_7.0.6.1/base-7.0.6.1/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[2]: Leaving directory '/home/lyang/StreamDeviceHowTo/keithley6517TestApp/src'
make[1]: *** [/opt/epics_7.0.6.1/base-7.0.6.1/configure/RULES_DIRS:85: src.install] Error 2
make[1]: Leaving directory '/home/lyang/StreamDeviceHowTo/keithley6517TestApp'
make: *** [/opt/epics_7.0.6.1/base-7.0.6.1/configure/RULES_DIRS:85: keithley6517TestApp.install] Error 2

MarkRivers, 2022-10-26:
I believe that means that you build Stream with the “calc” module, so you need to build the test application with “calc” as well.  Add it to /home/lyang/StreamDeviceHowTo/configure/RELEASE.

LeeYangLBLBCS, 2022-10-27:
```
That didn't help. Same error with added calc to the RELEASE.
I then tried an older epics base (7.0.4), the compilation succeeded.
But when I tried to run the IOC, I got an error:
**Declaration of recordtype(scalcout) preceeded full definition.
Error at or before "}" in file "dbd/HPE3631Atest.dbd" line 3151** (dbd file attached).

yang@leeVBUB20:~/StreamDeviceTest/HPE3631A/iocBoot/iocHPE3631Atest$ ../../bin/linux-x86_64/HPE3631Atest st.cmd
#!../../bin/linux-x86_64/HPE3631Atest
< envPaths
epicsEnvSet("IOC","iocHPE3631Atest")
epicsEnvSet("TOP","/home/lyang/StreamDeviceTest/HPE3631A")
epicsEnvSet("ASYN","/opt/epics/modules/synApps_6_1_epics7/support/asyn-R4-38")
epicsEnvSet("CALC","/opt/epics/modules/synApps_6_1_epics7/support/calc-R3-7-4")
epicsEnvSet("STREAM","/opt/epics/modules/synApps_6_1_epics7/support/StreamDevice-2-8-23")
epicsEnvSet("EPICS_BASE","/opt/epics/base-7.0.4")
cd "/home/lyang/StreamDeviceTest/HPE3631A"
## Register all support components
dbLoadDatabase "dbd/HPE3631Atest.dbd"
**Declaration of recordtype(scalcout) preceeded full definition.
Error at or before "}" in file "dbd/HPE3631Atest.dbd" line 3151**
HPE3631Atest_registerRecordDeviceDriver pdbbase
registerRecordDeviceDriver failed aSub
registerRecordDeviceDriver failed aai
registerRecordDeviceDriver failed aao
registerRecordDeviceDriver failed ai
registerRecordDeviceDriver failed ao
registerRecordDeviceDriver failed bi
registerRecordDeviceDriver failed bo
registerRecordDeviceDriver failed calc
registerRecordDeviceDriver failed calcout
registerRecordDeviceDriver failed compress
registerRecordDeviceDriver failed dfanout
registerRecordDeviceDriver failed fanout
registerRecordDeviceDriver failed histogram
registerRecordDeviceDriver failed int64in
registerRecordDeviceDriver failed int64out
registerRecordDeviceDriver failed longout
registerRecordDeviceDriver failed lsi
registerRecordDeviceDriver failed lso
registerRecordDeviceDriver failed mbbi
registerRecordDeviceDriver failed mbbiDirect
registerRecordDeviceDriver failed mbbo
registerRecordDeviceDriver failed permissive
registerRecordDeviceDriver failed printf
registerRecordDeviceDriver failed sel
registerRecordDeviceDriver failed state
registerRecordDeviceDriver failed stringin
registerRecordDeviceDriver failed stringout
registerRecordDeviceDriver failed waveform
registryJLinkAdd failed calc
registryJLinkAdd failed const
registryJLinkAdd failed debug
registryJLinkAdd failed state
registryJLinkAdd failed trace
## Load record instances
#dbLoadRecords("db/xxx.db","user=lyang")
cd "/home/lyang/StreamDeviceTest/HPE3631A/iocBoot/iocHPE3631Atest"
iocInit
Starting iocInit
############################################################################
## EPICS R7.0.4
## Rev. 2022-10-14T06:44-0700
############################################################################
iocRun: All initialization complete
## Start any sequence programs
#seq sncxxx,"user=lyang"
epics> 
[HPE3631Atest.dbd.txt](https://github.com/epics-modules/asyn/files/9874583/HPE3631Atest.dbd.txt)

MarkRivers, 2022-10-27:
Because you built stream with calc you need to include the calc dbd and lib files in all applications you build with stream, including this one.

closed 2022-11-09 by LeeYangLBLBCS