epics-modules/asyn #167: Is further support for AAI and AAO records pLanned?

ericonr, 2023-01-11:
Hi! Thanks to #162 being merged (great work there!), asyn now supports the AAI and AAO record types natively. However, the support is equivalent to the Waveform record, and doesn't address things like allocating the underlying storage for the AAI/O records, which is a pretty interesting feature, imo.

Do you think future asyn development plans to address that and enable programmers further control over AAI/O records?

MarkRivers, 2023-01-11:
I would welcome insights on how the asyn device support could implement the ability to allocate the storage for the record. It seems to me that to take advantage of that feature the device support would need to get the pointer from the underlying driver. But I don’t think there is an asyn interface to communicate that information?

ericonr, 2023-01-13:
Agreed, I think a new interface would be necessary...

Something like

```cpp
createParam(P_ArrayString, asynParamFloat64Array, &P_Array);
setFloat64ArrayStorage(P_Array, pointer, size); // this could mean adding arrays to the paramLibrary or having a separate helper class
```

might work? And then we'd have to decide if it makes sense to support changing the array storage while it's running or if it's an operation that only works before the record is initialized...

ericonr, 2023-01-13:
Beyond that, there's also a concern about how to "synchronize" operations on the array, so that any CA clients have a consistent view of the array...

This is something about the AAI and AAO records that isn't clear to me, actually. Once the array has been allocated, can a program manipulate them freely (or with some basic locking function)? As in:

- write new values into the AAI array and then somehow create a notification about changes
- read updated values from a write to the PV