epics-modules/asyn #172: asyn ports cannot be destroyed on IOC shutdown

exzombie, 2023-03-17:
When the IOC terminates, the ports are not given a chance to clean up and release resources. Some resources, like memory, are not problematic: the OS will reclaim the memory of the terminated process. But there are more problematic resources, such as improperly closed connections to devices, that can consume resources on devices themselves.

I have attempted to work around this in my driver by registering a function with `epicsAtExit()` which deletes the driver, allowing the destructor to clean up. This causes a crash because EPICS does not stop scan threads on IOC exit. This necessitates disabling the port through `asynManager` before destroying the driver. The IOC no longer crashes on shutdown, but it does print spurious error messages when the scan threads poke the disabled port.

This workaround works well for our use case. It is not very "clean", however, because it leaves dangling references and simply trusts that they will not be used anymore.

Other users have [noticed this behavior](https://epics.anl.gov/tech-talk/2022/msg00990.php). The above workaround was proposed and discussed in the linked thread, and steps towards a better solution were outlined by Michael Davidsaver.

daykin, 2023-03-21:
+1 to this being a nice improvement. It would make troubleshooting serious memory leaks in our drivers much easier if we can work towards knowing that the release branches of base, asyn, ADCore, etc. are valgrind memcheck/leak-check  clean, without having to sift through tons of "leaks" that are actually just a negligible fixed cost that eventually gets freed up by the OS. 

Plus, leaving cleanup work to the OS can mess up some unit test frameworks that destroy and rebuild the same objects for different tests.

Let me know if I can assist in any way.

exzombie, 2025-12-03:
The PR was merged.

closed 2025-12-03 by exzombie