epics-modules/asyn #196: Build fails on Window when TIRPC is YES

hanak, 2023-10-30:
Problem: We are building for both Linux and Windows. When TIRPC is YES, testGpibApp fails to build.

Cause: While "asyn" Makefile states:
```
ifeq ($(TIRPC),YES)
  USR_INCLUDES_Linux += -I/usr/include/tirpc
  asyn_SYS_LIBS_Linux += tirpc
endif
```
"testGpibApp" Makefile states:
```
ifeq ($(TIRPC),YES)
  USR_INCLUDES += -I/usr/include/tirpc
  testGpib_SYS_LIBS += tirpc
endif
```
Since the suffix `_Linux` is missing, the result applies to all OS.

Workaround: Set `USR_INCLUDES_Linux`, `asyn_SYS_LIBS_Linux `, and `testGpib_SYS_LIBS` in `CONFIG_SITE` instead of `TIRPC`.

ericonr, 2025-01-07:
@hanak can you open a PR changing `testGpibApp` to use the `_Linux` suffix? That seems like the most reasonable fix, and might have been missed back when TIRPC support was first added.

MarkRivers, 2025-01-07:
The problem is probably because you are modifying CONFIG_SITE to set TIRPC=YES.  

If you do that, then you need to have a  CONFIG_SITE.Common.windows-x64 that sets TIRPC=NO.

Preferably, don't change CONFIG_SITE. but rather add TIRPC=YES to CONFIG_SITE.Common.linux-x86_64.  I just added that as the default in that file

hanak, 2025-01-08:
@MarkRivers Thank you. Still, there is a difference between "asyn" Makefile and "testBpibApp" Makefile which causes different behavior while compiling on Windows. I can make it disappear by using CONFIG_SITE dedicated to a specific OS. Yet, is this meant to be like this?