epics-modules/asyn #201: drvAsynFTDIPortConfigure Hex support

matt-shawsda, 2024-01-25:
Hi,
I'm trying to use the drvAsynFTDIPortConfigure to connect to a usb device for streamdevice.
The devices vendor ID is a hex value, and as this has a non integer value I am getting the error:

st.cmd line 26: Illegal integer 'f0fa'

Can anyone advise. I have tried inputting this as its decimal and that wouldn't connect

Cheers

norumwe12, 2024-01-25:
The IOC shell accepts C-style integers.  You must precede the ‘f0fa’ with ‘0x’ to indicate that the value is hexadecimal.

> Hi,
> I'm trying to use the drvAsynFTDIPortConfigure to connect to a usb device for streamdevice.
> The devices vendor ID is a hex value, and as this has a non integer value I am getting the error: [...]

— 
Eric Norum
***@***.***

norumwe12, 2024-01-25:
Hmm.  I see now that you tried decimal (61690).  What value did you use for the product ID?  Has the FTDI chip been programmed with different firmware?  The data sheets that I have handy (FT232H and FT4232H) indicate a vendor code of 0x0403.

matt-shawsda, 2024-01-26:
Im using vendor code 0x0403 and product id 0xf0fa which is a thorlabs laser flipper 

My ioc is crashing with a segmentation fault core dump

closed 2024-01-29 by matt-shawsda