epics-modules/asyn #204: Need help on this error message

km-b214, 2024-02-23:
As a newbie in EPICS. I would like to setup the device, keller, based on EPICS. However, after setting up the RELEASE to match the system directory, then make. It shows the following message that keeps continue forever. 

I wondering if anyone can guide me to fix it. Thanks. 

christ@christ-vm:~/EPICS/keller$ make
make -C ./configure install
make[1]: Entering directory '/home/kritsada/EPICS/keller/configure'
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[2]: Entering directory '/home/kritsada/EPICS/keller/configure/O.linux-x86_64'
perl -CSD /home/kritsada/EPICS/epics-base/bin/linux-x86_64/convertRelease.pl checkRelease
make[2]: Leaving directory '/home/kritsada/EPICS/keller/configure/O.linux-x86_64'
make[1]: Leaving directory '/home/kritsada/EPICS/keller/configure'
make -C ./Keller3040App install
make[1]: Entering directory '/home/kritsada/EPICS/keller/Keller3040App'
make -C ./src install
make[2]: Entering directory '/home/kritsada/EPICS/keller/Keller3040App/src'
make -C O.linux-x86_64 -f ../Makefile TOP=../../.. \
    T_A=linux-x86_64 install
make[3]: Entering directory '/home/kritsada/EPICS/keller/Keller3040App/src/O.linux-x86_64'
Creating dbd file Keller3040.dbd
perl -CSD /home/kritsada/EPICS/epics-base/bin/linux-x86_64/dbdExpand.pl   -I. -I.. -I../O.Common -I../../../dbd -I/home/kritsada//EPICS/support/asyn/dbd -I/home/kritsada//EPICS/support/modbus/dbd -I/home/kritsada/EPICS/epics-base/dbd -o Keller3040.dbd base.dbd asyn.dbd drvAsynSerialPort.dbd modbusSupport.dbd
perl -CSD /home/kritsada/EPICS/epics-base/bin/linux-x86_64/registerRecordDeviceDriver.pl   -I. -I.. -I../O.Common -I../../../dbd -I/home/kritsada//EPICS/support/asyn/dbd -I/home/kritsada//EPICS/support/modbus/dbd -I/home/kritsada/EPICS/epics-base/dbd    -o Keller3040_registerRecordDeviceDriver.cpp \
            ../O.Common/Keller3040.dbd Keller3040_registerRecordDeviceDriver /home/kritsada/EPICS/keller
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include         -I/home/kritsada//EPICS/support/asyn/include   -I/home/kritsada//EPICS/support/modbus/include -I/home/kritsada/EPICS/epics-base/include/compiler/gcc -I/home/kritsada/EPICS/epics-base/include/os/Linux -I/home/kritsada/EPICS/epics-base/include        -MM -MF Keller3040_registerRecordDeviceDriver.d  Keller3040_registerRecordDeviceDriver.cpp
Creating dbd file Keller3040.dbd
perl -CSD /home/kritsada/EPICS/epics-base/bin/linux-x86_64/dbdExpand.pl   -I. -I.. -I../O.Common -I../../../dbd -I/home/kritsada//EPICS/support/asyn/dbd -I/home/kritsada//EPICS/support/modbus/dbd -I/home/kritsada/EPICS/epics-base/dbd -o Keller3040.dbd base.dbd asyn.dbd drvAsynSerialPort.dbd modbusSupport.dbd
perl -CSD /home/kritsada/EPICS/epics-base/bin/linux-x86_64/registerRecordDeviceDriver.pl   -I. -I.. -I../O.Common -I../../../dbd -I/home/kritsada//EPICS/support/asyn/dbd -I/home/kritsada//EPICS/support/modbus/dbd -I/home/kritsada/EPICS/epics-base/dbd    -o Keller3040_registerRecordDeviceDriver.cpp \
            ../O.Common/Keller3040.dbd Keller3040_registerRecordDeviceDriver /home/kritsada/EPICS/keller
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE          -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall      -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include         -I/home/kritsada//EPICS/support/asyn/include   -I/home/kritsada//EPICS/support/modbus/include -I/home/kritsada/EPICS/epics-base/include/compiler/gcc -I/home/kritsada/EPICS/epics-base/include/os/Linux -I/home/kritsada/EPICS/epics-base/include        -MM -MF Keller3040_registerRecordDeviceDriver.d  Keller3040_registerRecordDeviceDriver.cpp
Creating dbd file Keller3040.dbd

km-b214, 2024-02-23:
Yes it solves the problem. Thanks.

MarkRivers, 2024-02-24:
The problem was solved.

closed 2024-02-24 by MarkRivers