epics-modules/asyn #207: mbboDirect with asyn:READBACK is not processed

exzombie, 2024-06-10:
With the database below, combined with a driver that periodically calls `setUIntDigitalParam()` and `callParamCallbacks()` for the `RANDOM` reason, the first record is processed, while the second is not. If it's relevant, I tried setting both `valueMask` and `interruptMask` parameters to `setUIntDigitalParam()` to `0xffff`.

```
record(longout, "$(user):random1") {
    field(DTYP, "asynUInt32Digital")
    field(OUT, "@asynMask($(port),0,0xffff) RANDOM")
    info(asyn:READBACK, 1)
}

record(mbboDirect, "$(user):random2") {
    field(DTYP, "asynUInt32Digital")
    field(OUT, "@asynMask($(port),0,0xffff) RANDOM")
    info(asyn:READBACK, 1)
}
```

exzombie, 2024-06-11:
Correction, the record `$(user):random2` *is* processed, but VAL remains unchanged. The change ends up in RVAL and is not propagated to VAL and B0-B1F.

MarkRivers, 2024-06-11:
It appears that the same block of code that sets RVAL also then sets VAL.
https://github.com/epics-modules/asyn/blob/33a66b58732521ee176b28dfeb9ebc1e2e985e47/asyn/devEpics/devAsynUInt32Digital.c#L1088

Can you add some debugging there to see why VAL is not changing?

exzombie, 2024-06-11:
What happens there is that RVAL is set from the callback result, then VAL is set, bit by bit, based on values of B1-B1F. RVAL is not considered, and bits are zero, so VAL stays at zero. Given that this piece of code is intended specifically for handling callbacks, that should be turned around, no? VAL should be set from RVAL (after a shift), and B1-B1F should be set from VAL.

PR #208 (2024-06-14): Fix mbboDirect asyn:READBACK
This change makes mbboDirect behave the same as mbbiDirect: RVAL is copied to VAL (shifted, if neccessary), then each of the fields representing individual bits is set as well.

Fixes #207
closed 2024-06-14 by MarkRivers