epics-modules/asyn #209: relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used

PeiyuQuan, 2024-07-17:
Hello
I didn't make any modification, just uninstall asyn and re-install asyn. I got this error.

It is Ubuntu 22.04. Asyn is the latest version.

**************************************************************************************************************************
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libtirpc.a(libtirpc_la-clnt_generic.o): warning: relocation against `__rpc_minfd' in read-only section `.text'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libtirpc.a(libtirpc_la-clnt_perror.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
*********************************************************************************************************************************

Part of installed result:

t -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include      -I/home/quan/epics/support/ipac/include   -I/home/quan/epics/support/seq/include   -I/home/quan/epics/support/calc/include   -I/home/quan/epics/support/sscan/include -I/home/quan/epics/base/include/compiler/gcc -I/home/quan/epics/base/include/os/Linux -I/home/quan/epics/base/include    -I/usr/include/tirpc    -c ../../asyn/vxi11/TDS3000Reboot.c
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_CALCMOD -DHAVE_DEVINT64  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include      -I/home/quan/epics/support/ipac/include   -I/home/quan/epics/support/seq/include   -I/home/quan/epics/support/calc/include   -I/home/quan/epics/support/sscan/include -I/home/quan/epics/base/include/compiler/gcc -I/home/quan/epics/base/include/os/Linux -I/home/quan/epics/base/include    -I/usr/include/tirpc    -c ../../asyn/drvPrologixGPIB/drvPrologixGPIB.c
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_CALCMOD -DHAVE_DEVINT64  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include      -I/home/quan/epics/support/ipac/include   -I/home/quan/epics/support/seq/include   -I/home/quan/epics/support/calc/include   -I/home/quan/epics/support/sscan/include -I/home/quan/epics/base/include/compiler/gcc -I/home/quan/epics/base/include/os/Linux -I/home/quan/epics/base/include    -I/usr/include/tirpc    -c ../../asyn/devGpib/devCommonGpib.c
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_CALCMOD -DHAVE_DEVINT64  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include      -I/home/quan/epics/support/ipac/include   -I/home/quan/epics/support/seq/include   -I/home/quan/epics/support/calc/include   -I/home/quan/epics/support/sscan/include -I/home/quan/epics/base/include/compiler/gcc -I/home/quan/epics/base/include/os/Linux -I/home/quan/epics/base/include    -I/usr/include/tirpc    -c ../../asyn/devGpib/devSupportGpib.c
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_CALCMOD -DHAVE_DEVINT64  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include      -I/home/quan/epics/support/ipac/include   -I/home/quan/epics/support/seq/include   -I/home/quan/epics/support/calc/include   -I/home/quan/epics/support/sscan/include -I/home/quan/epics/base/include/compiler/gcc -I/home/quan/epics/base/include/os/Linux -I/home/quan/epics/base/include    -I/usr/include/tirpc    -c ../../asyn/devGpib/boSRQonOff.c
/usr/bin/ar -rc    libasyn.a    asynManager.o epicsInterruptibleSyscall.o asynGpib.o drvAsynSerialPort.o drvAsynIPPort.o drvAsynIPServerPort.o asynInt32Base.o asynInt32SyncIO.o asynInt64Base.o asynInt64SyncIO.o asynInt8ArrayBase.o asynInt8ArraySyncIO.o asynInt16ArrayBase.o asynInt16ArraySyncIO.o asynInt32ArrayBase.o asynInt32ArraySyncIO.o asynInt64ArrayBase.o asynInt64ArraySyncIO.o asynUInt32DigitalBase.o asynUInt32DigitalSyncIO.o asynFloat64Base.o asynFloat64SyncIO.o asynFloat32ArrayBase.o asynFloat32ArraySyncIO.o asynFloat64ArrayBase.o asynFloat64ArraySyncIO.o asynOctetBase.o asynOctetSyncIO.o asynGenericPointerBase.o asynGenericPointerSyncIO.o asynEnumBase.o asynEnumSyncIO.o asynCommonSyncIO.o asynOptionSyncIO.o asynStandardInterfacesBase.o asynShellCommands.o asynInterposeCom.o asynInterposeEos.o asynInterposeFlush.o asynInterposeDelay.o asynInterposeEcho.o ParamListInvalidIndex.o ParamListParamNotFound.o ParamValNotDefined.o ParamValStringSizeRequestTooBig.o ParamValWrongType.o ParamValValueNotChanged.o paramVal.o asynPortDriver.o asynPortClient.o devAsynOctet.o asynEpicsUtils.o devAsynInt32.o devAsynInt32TimeSeries.o devAsynUInt32Digital.o devAsynFloat64.o devAsynXXXArray.o devAsynFloat64TimeSeries.o devEpicsPvt.o devAsynInt64TimeSeries.o devAsynInt64.o asynRecord.o drvAsyn.o vxi11core_xdr.o drvVxi11.o E5810Reboot.o E2050Reboot.o TDS3000Reboot.o drvPrologixGPIB.o devCommonGpib.o devSupportGpib.o boSRQonOff.o 
/usr/bin/ranlib libasyn.a
/usr/bin/g++ -o libasyn.so  -shared -fPIC -Wl,-hlibasyn.so -L/home/quan/epics/support/asyn/lib/linux-x86_64 -L/home/quan/epics/base/lib/linux-x86_64 -Wl,-rpath,/home/quan/epics/support/asyn/lib/linux-x86_64 -Wl,-rpath,/home/quan/epics/base/lib/linux-x86_64          -rdynamic -m64         asynManager.o epicsInterruptibleSyscall.o asynGpib.o drvAsynSerialPort.o drvAsynIPPort.o drvAsynIPServerPort.o asynInt32Base.o asynInt32SyncIO.o asynInt64Base.o asynInt64SyncIO.o asynInt8ArrayBase.o asynInt8ArraySyncIO.o asynInt16ArrayBase.o asynInt16ArraySyncIO.o asynInt32ArrayBase.o asynInt32ArraySyncIO.o asynInt64ArrayB [... cut ...]
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/quan/epics/base/configure/RULES_BUILD:316: libasyn.so] Error 1
make[2]: Leaving directory '/home/quan/epics/support/asyn/asyn/O.linux-x86_64'
make[1]: *** [/home/quan/epics/base/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/quan/epics/support/asyn/asyn'
make: *** [/home/quan/epics/base/configure/RULES_DIRS:85: asyn.install] Error 2
quan@PC98545:~/epics/support/asyn$

MarkRivers, 2024-07-17:
What package did you install for tirpc? You seem to have only libtirpc.a, not the needed .so file.

Mark

PeiyuQuan, 2024-07-17:
Mark
I installed tirpc two years ago. I forget what version it is. But it worked before until last week.
It has libtirpc.a, but their path are different. Why the path is  /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libtirpc.a during installing.
quan@PC98545:~/epics/support/asyn$ apt-file find libtirpc.a
libtirpc-dev: /usr/lib/x86_64-linux-gnu/libtirpc.a
********************************************************************************************************************************************
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libtirpc.a(libtirpc_la-clnt_generic.o): warning: relocation against __rpc_minfd' in read-only section .text'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libtirpc.a(libtirpc_la-clnt_perror.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
***************************************************************************************************************************************

MarkRivers, 2024-07-18:
I’m on vacation. I can look at this tomorrow.

You need to see if libtirpc.so is present on your system. That is what you need.

Mark

PeiyuQuan, 2024-07-18:
Mark
Yes. They are there, Have a safety trip, No Rush.
quan@PC98545:~/epics/support/asyn$ apt-file find libtirpc.so
libtirpc-dev: /usr/lib/x86_64-linux-gnu/libtirpc.so
libtirpc3: /lib/x86_64-linux-gnu/libtirpc.so.3
libtirpc3: /lib/x86_64-linux-gnu/libtirpc.so.3.0.0

MarkRivers, 2024-07-19:
What versions of these are you using:
- asyn
- EPICS base
- Linux

On my RHEL9 system there is no libtirpc.a, just these .so files:
```
(base) [epics@corvette ~]$ find /usr -xdev -name 'libtirpc.so*'
/usr/lib64/libtirpc.so.3
/usr/lib64/libtirpc.so.3.0.0
/usr/lib64/libtirpc.so

(base) [epics@corvette ~]$ ls -l /usr/lib64/libtirpc*
lrwxrwxrwx. 1 root root     17 Mar 21 14:16 /usr/lib64/libtirpc.so -> libtirpc.so.3.0.0
lrwxrwxrwx. 1 root root     17 Mar 21 14:16 /usr/lib64/libtirpc.so.3 -> libtirpc.so.3.0.0
-rwxr-xr-x. 1 root root 196248 Mar 21 14:16 /usr/lib64/libtirpc.so.3.0.0
```

I suspect something else has changed on your system besides just uninstalling and re-installing asyn.

It seems like it is trying to link against libtirpc.a rather than libtirpc.so, but I am not sure.

PeiyuQuan, 2024-07-19:
Mark

The Base is 7.07. Ubuntu 22.04.4 LTS, Asyn is the latest version I download it this week. 
I don't think it is EPICS issue, i think some documents in my PC is changed. I planed to re-install system. 
GLIBC is ldd (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35. But the error report show "stderr@@GLIBC_2.2.5' can not be used".

quan@PC98545:~$ sudo apt-file find libtirpc*
[sudo] password for quan: 
libtirpc-dev: /usr/lib/x86_64-linux-gnu/libtirpc.so
libtirpc3: /lib/x86_64-linux-gnu/libtirpc.so.3
libtirpc3: /lib/x86_64-linux-gnu/libtirpc.so.3.0.0

quan@PC98545:~/epics/support/asyn$ ls -l /lib/x86_64-linux-gnu/libtirpc*
-rw-r--r-- 1 root root 404664 Jul 25  2022 /lib/x86_64-linux-gnu/libtirpc.a
lrwxrwxrwx 1 root root     17 Jul 25  2022 /lib/x86_64-linux-gnu/libtirpc.so.3 -> libtirpc.so.3.0.0
-rw-r--r-- 1 root root 182912 Jul 25  2022 /lib/x86_64-linux-gnu/libtirpc.so.3.0.0

PeiyuQuan, 2024-07-19:
Mark
I thought I copied libtirpc.so last week and I have an incorrect operation. I can't remember clearly. I will try to fix it. 
Thank you
Peiyu Quan

MarkRivers, 2024-07-19:
> GLIBC is ldd (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35. But the error report show "stderr@@GLIBC_2.2.5' can not be used".

That is normal, it is not a reason to re-install.  My system is RHEL 9.4, gcc 11.4.1, GLIBC 2.34.  This is what I see when I look at the GLIBC functions that libasyn.so is linked with:
```
(base) [epics@corvette linux-x86_64]$ nm libasyn.so  | grep GLIBC | grep -v GLIBCXX
                 U bind@GLIBC_2.2.5
                 U calloc@GLIBC_2.2.5
                 U cfsetispeed@GLIBC_2.2.5
                 U cfsetospeed@GLIBC_2.2.5
                 U close@GLIBC_2.2.5
                 U connect@GLIBC_2.2.5
                 U __ctype_b_loc@GLIBC_2.3
                 w __cxa_finalize@GLIBC_2.2.5
                 U __errno_location@GLIBC_2.2.5
                 U fclose@GLIBC_2.2.5
                 U fcntl@GLIBC_2.2.5
                 U fflush@GLIBC_2.2.5
                 U fopen@GLIBC_2.2.5
                 U fprintf@GLIBC_2.2.5
                 U fputc@GLIBC_2.2.5
                 U free@GLIBC_2.2.5
                 U getpeername@GLIBC_2.2.5
                 U getsockname@GLIBC_2.2.5
                 U ioctl@GLIBC_2.2.5
                 U __isoc99_sscanf@GLIBC_2.7
                 U listen@GLIBC_2.2.5
                 U malloc@GLIBC_2.2.5
                 U memchr@GLIBC_2.2.5
                 U memcmp@GLIBC_2.2.5
                 U memcpy@GLIBC_2.14
                 U memmove@GLIBC_2.2.5
                 U memset@GLIBC_2.2.5
                 U open@GLIBC_2.2.5
                 U poll@GLIBC_2.2.5
                 U printf@GLIBC_2.2.5
                 U puts@GLIBC_2.2.5
                 U read@GLIBC_2.2.5
                 U realloc@GLIBC_2.2.5
                 U recvfrom@GLIBC_2.2.5
                 U recv@GLIBC_2.2.5
                 U send@GLIBC_2.2.5
                 U sendto@GLIBC_2.2.5
                 U setsockopt@GLIBC_2.2.5
                 U shutdown@GLIBC_2.2.5
                 U sprintf@GLIBC_2.2.5
                 U stpcpy@GLIBC_2.2.5
                 U strchr@GLIBC_2.2.5
                 U strcmp@GLIBC_2.2.5
                 U strcpy@GLIBC_2.2.5
                 U strerror@GLIBC_2.2.5
                 U strlen@GLIBC_2.2.5
                 U strncmp@GLIBC_2.2.5
                 U strncpy@GLIBC_2.2.5
                 U strrchr@GLIBC_2.2.5
                 U strstr@GLIBC_2.2.5
                 U strtod@GLIBC_2.2.5
                 U strtol@GLIBC_2.2.5
                 U strtoul@GLIBC_2.2.5
                 U tcdrain@GLIBC_2.2.5
                 U tcflush@GLIBC_2.2.5
                 U tcsendbreak@GLIBC_2.2.5
                 U tcsetattr@GLIBC_2.2.5
                 U vfprintf@GLIBC_2.2.5
                 U write@GLIBC_2.2.5
```
Note that most of the functions are using the version from 2.2.5, but some use 2.3, 2.7, 2.14 etc.

> I thought I copied libtirpc.so last week and I have an incorrect operation. I can't remember clearly. I will try to fix it.

You should never copy system libraries in an attempt to fix a problem you don't understand.  It will just lead to total confusion.

PeiyuQuan, 2024-07-20:
Hello Mark
I lost my libtirpc.so. I have created a new one in /lib/x86_64-linux-gnu/. asyn driver has been installed. Thank you so much.

Peiyu

closed 2024-11-05 by MarkRivers