epics-modules/asyn #212: CI builds are failing because of rpc.h not found

MarkRivers, 2025-01-07:
I just pushed a commit to asyn/configure/CONFIG_SITE.Common.linux-x86_64 to add TIPRC=YES.  This should save people trouble, because that line is commented out by default in CONFIG_SITE.  If it is uncommented in CONFIG_SITE so Linux builds succeed, then Windows builds will fail.

However, I found the CI builds are failing.  I don't understand this, since one of the first steps in the build is this:
```
Run echo TIRPC=YES > configure/CONFIG_SITE.Common.linux-x86_64
```
That replaces the file I just modified, so my change should have no effect.

However, the Linux builds all fail like this:
```
 gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -fdiagnostics-color -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DBUILDING_asyn_API  -D_X86_64_ -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -DUNIX  -Dlinux      -O3 -g   -Wall -Werror-implicit-function-declaration    -DUSE_TYPED_RSET -DUSE_TYPED_DSET -DUSE_TYPED_DRVET -DHAVE_LSREC -DHAVE_CALCMOD -DHAVE_DEVINT64  -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I../../asyn/drvAsynSerial/os/Linux -I../../asyn/drvAsynSerial/os/default -I.. -I../../asyn/asynDriver -I../../asyn/asynGpib -I../../asyn/drvAsynSerial -I../../asyn/interfaces -I../../asyn/miscellaneous -I../../asyn/asynPortDriver/exceptions -I../../asyn/asynPortDriver -I../../asyn/asynPortClient -I../../asyn/devEpics -I../../asyn/asynRecord -I../../asyn/vxi11 -I../../asyn/drvPrologixGPIB -I../../asyn/gsIP488 -I../../asyn/ni1014 -I../../asyn/devGpib -I../../include/compiler/gcc -I../../include/os/Linux -I../../include    -I/home/runner/.cache/base-7.0/include/compiler/gcc -I/home/runner/.cache/base-7.0/include/os/Linux -I/home/runner/.cache/base-7.0/include   -I/home/runner/.cache/seq-R2-2-8/include   -I/home/runner/.cache/sscan-R2-11-5/include   -I/home/runner/.cache/calc-R3-7-4/include   -I/home/runner/.cache/ipac-2.16/include    -I/usr/include/tirpc    -c vxi11core_xdr.c
  make[2]: Leaving directory '/home/runner/work/asyn/asyn/asyn/O.linux-x86_64'
  vxi11core.h:9:10: fatal error: rpc/rpc.h: No such file or directory
      9 | #include <rpc/rpc.h>
        |          ^~~~~~~~~~~
  compilation terminated.
  make[2]: *** [/home/runner/.cache/base-7.0/configure/RULES_BUILD:259: vxi11core_xdr.o] Error 1
  make[2]: *** Waiting for unfinished jobs....
  make[2]: Entering directory '/home/runner/work/asyn/asyn/asyn/O.linux-x86_64'
```
Note that the compilation command contains this:
```
-I/usr/include/tirpc 
```
It is failing because rpc/rpc.h cannot be found.  But that should be found in /usr/include/tirpc.

Is tiprc not being installed for some reason?

MarkRivers, 2025-01-13:
@ralphlange can I ask you to look at this?  It seems like the tirpc package is not being installed.  I can't find any mention of tirpc in the asyn .ci or .ci-local directory.  What is supposed to be installing the tirpc package?

ralphlange, 2025-01-14:
Before your last commit, all builds (Linux and Windows) were succeeding.

As you pointed out, TIRPC was set to yes as part of the CI job.
https://github.com/epics-modules/asyn/blob/0ed37b298307f3c32718bd6b28fcdf4bd459bacf/.github/workflows/ci-scripts.yml#L111

If you look at the builds before your commit, `-I/usr/include/tirpc` was set on the compile line and the headers were found. The library was installed on all the runners.

If you look closer at the top of the logs, you will find that GitHub switched the `ubuntu-latest` image (that the ASYN builds use) from version 22.04 to 24.04 a few days ago.
Obviously, the new Ubuntu-24.04 runner image does not have `libtirpc-dev` installed. I will add that to the configuration and create a PR.

MarkRivers, 2025-01-14:
> Obviously, the new Ubuntu-24.04 runner image does not have libtirpc-dev installed. I will add that to the configuration and create a PR.

Thanks.  My uncertainty was whether libtirpc-devel was really previously installed by default on ubuntu, or whether it was being installed by the CI tools in ways I couldn't figure out.

MarkRivers, 2025-02-12:
This has been fixed.

closed 2025-02-12 by MarkRivers