epics-modules/asyn #213: Update CI build matrix

ralphlange, 2025-01-14:
# The CI build (using GitHub Actions) lacks maintenance.

I am willing to update things as a one-off project, part of my current CI related PR.

@FreddieAkeroyd @exzombie @MarkRivers @mdavidsaver @keenanlang 
But: You must help and tell me which combinations of build parameters the CI builds should cover.
Obviously, running all possible combinations creates way too many jobs. What we need is a useful, sensible subset.

## Available options (not all shown)

_(Everything italics and in brackets)_ is currently not used

### Platforms (native)
_(Ubuntu 20)_
_(Ubuntu 22)_
Ubuntu 24

Windows-2019
_(Windows-2022)_
_(Windows-2025)_

_(MacOS 13)_
_(MacOS 14)_

### Platforms (cross from Linux to)
RTEMS-4.9
RTEMS-4.10
_(RTEMS-5.1)_
_(Linux@ARM64)_
_(Windows/MinGW)_

### Platforms (using docker)
_(CentOS:8)_
_(RockyLinux:9)_
_(Fedora:33)_
_(Fedora:latest)_

### Sets of dependency versions
_(**dep1**: SNCSEQ 2.2.8, SSCAN 2.11.3, CALC 3.7.3, IPAC 2.15)_
**dep2**: SNCSEQ 2.2.8, SSCAN 2.11.5, CALC 3.7.4, IPAC 2.16
_(**dep-latest**: default branch on all)_

### Configurations
shared (default)
_(debug)_
static (not on MacOS)
_(static-debug (not on MacOS))_

### Compilers (all available in different versions)
GNU (not on MacOS)
_(clang)_
MSVC (only Windows)

### Special cases
_(Only use EPICS build system and libcom (no device support))_
_(Set INSTALL_LOCATION)_

tboegi, 2025-01-14:
@ralphlange  (hopefully this is useful information)
I am currently using Ubuntu 20:
https://github.com/EuropeanSpallationSource/m-epics-ethercatmc/blob/master/.github/workflows/ci-scripts-build.yml
However, that should be update towards Ubuntu 22.
Or, if nobody else is on Ubuntu 22, to Ubuntu 24.

FreddieAkeroyd, 2025-01-30:
Regarding configurations, static and shared probably needed for most, debug is useful on windows as it uses different libraries and so a CI spots linker issues. Probably do not need all combination with all compiler versions as linker requirements are similar e.g. just all three with VS2022 and then maybe just release or static with the others to check compiler. How many VS versions do we feel we need? Personally we are now using Windows server 2019 and 2022 with Visual Studio 2022 (server 2022 is based on windows 11 based whereas 2019 was Windows 10 LTSC). We don't use MinGW ourselves, but this is well used i understand.